cedricp/ddt4all #386: My ELM327 isn't found data and doesn't automatic search

Babareis41, 2020-05-17:
Hi,my ELM327 ( With USB Cable ) isn't found data and doesn't automatic search.I tried againly read data but it isn't solve.First,I think my ELM327 is Chinese clone but I can connect to my car's brain with Scanmaster.Then I don't think my ELM327 is clone.Can you help me please ? My Car Is Megane 4 Sedan.

cedricp, 2020-05-18:
DDT4All needs special ELM functions not used by scanmaster.

Babareis41, 2020-05-18:
I think my chip is pic18f25k80 because my friend has same elm327 and he use succesfully ddt4all

Babareis41, 2020-05-20:
Hey again,I have a problem : "0 loaded ecus" How I Can Solve It ?

djsweat, 2020-07-20:
Hi, please test the ELM device with the android app: ELM327 Identifier, if the results stops after 1.4 version this means the device is OK. Please verify if all commands from above 1.4 version are done, with green colour.

pakoto, 2020-07-20:
Last fake elm327 WIFI i had, shows all ok in elm327 identifier, but don´t work. Black epoxy chip inside instead PIC.

Now, "all elm327 wifi valid" is not true...

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif