cedricp/ddt4all #387: I Have A Problem : "0 Loaded Ecu's"

Babareis41, 2020-05-20:
Hi,I Have 0 Loaded Ecu's Problem.How Can I Solve It ?

Thanks.

tanerdogan, 2020-05-20:
Change ELM.

Babareis41, 2020-05-20:
Reis senin kullandığın ve çalışan varsa linkler misin

tanerdogan, 2020-05-20:
Senin ELM cihazın çakma çin malı değil dimi? Bundan eminsen ecu.zip indirmen lazım, bu program tek başına çalışmaz. ecu.zip link: https://yadi.sk/d/-oZOBfV03wTaoQ

Babareis41, 2020-05-20:
Cihazı iade ettim cihazdandır dediler.Senin kullandığın obd2 cihazı varsa ddt4all uyumlu linkini atabilir misin oradan alayım bir de

tanerdogan, 2020-05-20:
Ben 2 yıl önce almıştım, çoğu çin malı sahte. Deneyeceksin şansını başka net bi satıcı yoktur. Onlarda çinden çalışıyor diye alıyor, ne sattıklarını bile bilmiyorlar.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif