cedricp/ddt4all #397: Econometer activation: doesn´t work after shitting down the car

JABICHOCU, 2020-06-04:
Hello to all you members of this community.I´d like to share with you this problem I have in hope of finding some kind of solution.I´ve got a 2013 Renault Captur.I´ve been doing a couple of changes with DDT4ALL on my instruments cluster.I´ve successfully managed to activate the clock...but the econometer only works the moment I activate it. Once I shut down the car...it won´t appear again,even though it is "active" on cluster settings as you can see on the picture.I have to say I could only activate it with the engine running...with the engine off it won´t even appear.Any body can give me any clues on this???  I appreciate your help...thanx!!
This is my cluster config>
[https://ibb.co/RYRgJHM](url)

JABICHOCU, 2020-06-04:
Link to PIC doesn´t seem to work :(

pakoto, 2020-06-04:
It happens with more users.
My dacia dokker 2014 with ecu EDC17C42 euro5 and clio IV ph1 cluster, econometer works. 
-dokker 2016 euro6 and clio iv ph2 cluster, don´t work. 
-traffic 2019 dci with stock cluster don´t work 

I don´t know if newer clio iv versions cluster problem, or ecu problem.

JABICHOCU, 2020-06-04:
Thanx pakoto...But isn´t it strange that it works and once  you stop the  engine it doesnt??

tanerdogan, 2020-06-04:
I have 2015 Clio 4, tired before and never seen its worked... I always using ddt4all ignition on, but not tried car was running...

JABICHOCU, 2020-06-04:
With the engine on it works for me...but the moment you stop the engine and lock the car, it never appears again! Even though it is "active" in the settings...I can find no explanation for this

AndyIreland, 2020-06-04:
> With the engine on it works for me...but the moment you stop the engine and lock the car, it never appears again! Even though it is "active" in the settings...I can find no explanation for this

Hi there @JABICHOCU  - I think I read in a Renault Owners manual that you have to enable or disable the econometer through the radio unit. - in my Ph1 Mk4 clio it has a Medianav Evolution radio and there is definately no settings for econometer in its settings .... so I am guessing they are talking about the r-link radio that is fitted to some Mk4 clios and if that has a setting in the r-link to turn the econometer on or off? - so if you have a r-link in your car , make sure the econometer is set to 'on' or 'enabled' as well as the setting in DDT4all.
On a similar note - when I try PyREN on my clio 4 and the cluster tests with the indicator lights on my dash absolutely every light lights up in testing ... except the econometer bar , which leads me to think that I havent even got the required LED(S) in my dash behind the econometer bar .. which means that even if I had a r-link radio instead and set the econometer bar in the r-link radio settings my econometer bar would still not work!

JABICHOCU, 2020-06-05:
Hello Andy, thank you very much for your info.That eventually explains the fact that I can see the econometer as soon as I code it but never appear again when I  stop the engine! .Unfortunately I have Medianav, not R-link which is the one with all the eco-driving options so there´s no way to setup this from my unit.
I take this opportunity to thank you as well for all your videos and tutorials on this DDT4all subject, they came in really handy for me as an initiation on this,there´re not many English speakers involved in this project!

AndyIreland, 2020-06-05:
> Hello Andy, thank you very much for your info.That eventually explains the fact that I can see the econometer as soon as I code it but never appear again when I stop the engine! .Unfortunately I [...]

Thank you very much, am glad you have found the tutorials useful -

pakoto, 2020-06-05:
econometer cluster light works without r-link / medianav. Eco option into nav only shows fuel info using BIC-MUI module.

AndyIreland, 2020-06-05:
> econometer cluster light works without r-link / medianav. Eco option into nav only shows fuel info using BIC-MUI module.
> 
> Tested with plug and radio, medianav 1, evo and evo2.
> 
> https://www.youtube.com/watch?v=kZKUri_5KFc
> 
> Cluster clio iv 2013 from car without tpms and s.s. + dokker 2014 euro5.

Thanks for that. The Clio 4 manual says to enable it or disable it refer to radio instructions - but I have a Medianav evolution and although ambience / econometer bar is enabled in DDT4ALL I go through all the settings of the medianav evolution and there is definately no setting for turning the econometer on or off ... thats why I was thinking that when they said refer to radio settings to turn the econometer on or off they must have been talking in the settings inside of a r-link radio unit  instead ?

JABICHOCU, 2020-06-06:
> 
> 
> econometer cluster light works without r-link / medianav. Eco option into nav only shows fuel info using BIC-MUI module.
> 
> Tested with plug and radio, medianav 1, evo and evo2.
> 
> https://www.youtube.com/watch?v=kZKUri_5KFc
> 
> Cluster clio iv 2013 from car without tpms and s.s. + dokker 2014 euro5.

Thanx for that info,Pakoto..nevertheless according to the Italian Clio forum this econometer works in some cars but not in others, apparently no reason for explaining that!

BTW>¿Lo de desactivar la iluminación permanente también se puede hacer por DDT4ALL???

pakoto, 2020-06-06:
> Thanx for that info,Pakoto..nevertheless according to the Italian Clio forum this econometer works in some cars but not in others, apparently no reason for explaining that!
>
> BTW>¿Lo de desactivar la [...]

Yes. TDB - config generale 2 - day_backlight_present "without" and press Rheostat jour. You can revert anytime.

JABICHOCU, 2020-06-06:
They say most of the ones working are Clios with R-link...have some cases without it but no clue about what makes it work or not

AndyIreland, 2020-06-08:
I still think (although not confirm) that some clusters had the LED at the back of the econometer bar and some clusters were made without the LED included for the econometer bar. 
One day soon i may physically take out the whole cluster on my clio 4 and actually have a look to see if the LED is included on mine  - when I do the cluster LED test with PyREN absolutely every LED lights up on my cluster except for 1 and that is the econometer

JABICHOCU, 2020-06-09:
Nevermind, Andy .People who have this feature working claim it is not worth much, it´s almost impossible to follow the "eco-driving" conditions demanded so as to even see it on green!

pakoto, 2020-06-09:
Works for me. 1.5 dci 90cv. No green only when accelerate and brake quickly, or using lower gear at high rpms some seconds.

JABICHOCU, 2020-06-09:
Hello Pakoto, now that I see you here, how would you recommend me to try and de-activate 
Start -Stop on a 2013 Captur? 

I´ve been searching for options on these Ecus on my car but I´m afraid of making a big mistake:

Cluster_X98_X87_X82_V2.4_20121001T113250

T4_BCM_DDT2000_SW11_0_20130128T102714

EMM_EDISON_DDT2000_SW10_V2_20121220T154401

EDC17C42_1701_810_170A_400_V02_20150224T131230

pakoto, 2020-06-09:
Tested with Dokker 1.5 dci 90cv euro6 with T4_VS_BCM uch. I don´t know if works with your uch.

[image]

Now testing hidden alarm option. I´ll upload video if works.

JABICHOCU, 2020-06-09:
I see you have BCM with SW15 (more modern car) but mine is SW11. I have fear of bricking the module as in some cases happened trying to change the Light settings on lower versions...thanks for your help anyway

pakoto, 2020-06-09:
Capture from dokker 2016 s.s. 

My dokker 2014 euro5 without start stop uses T4 BCM SW11 and no have GEE screen. Are you sure is correct?

Si eres de España también que haces hablando ingles conmigo? XD

JABICHOCU, 2020-06-09:
I speak with you here because I can´t reach you on the Spanish Dacia forums! ㋛ I´m not logged there!

JABICHOCU, 2020-06-10:
Una última pregunta,¿tienes idea de porqué se me resetea la aguja del combustible cada vez que cambio algo en el cuadro??? Al codificar el reloj , la iluminación de esferas y todo esto siempre se me queda subida a tope,aunque el depósito lo tenga a medias

JABICHOCU, 2020-06-11:
Pakoto??? Where are you???

pakoto, 2020-06-11:
dacieros.forumfree.net o dacieros en youtube

When make changes and reset, gauge recalibrate. Also happens when disconnect battery or cluster from car.

JABICHOCU, 2020-06-12:
Thank you...now I understand

JABICHOCU, 2020-06-13:
Please, Pakoto...have a look at this! 
https://www.solocaptur.com/threads/colores-evo-en-para-medianav-1.2837/#post-30663

JABICHOCU, 2020-07-05:
PROBLEM SOLVED! This morning I tried coding it again.With the engine on, immediately I could see the leds in orange. Then I put the car INTO GEAR and as soon as I inserted second gear the whole thing started working...changing to green and back to orange.The other time I just didn´t move the car enough so as to let it work properly...now we know what happened! Thanx to you all

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif