cedricp/ddt4all #400: PyRen Demo shuts down on vers. 09p

AndyIreland, 2020-06-07:
hi there - when I try demo mode with vers. 09p and select 46 - Clio IV the window shuts down with 
[image]

so just to recap:
I select 09p
[image]

I select  DEMO then 46 Clio IV
[image]

then it exits to this screen:
[image]

@shrlnm  @cedricp  and any others that may be able to help please?

Marianpol, 2020-06-08:
It's normal, becouse you have selected demo mode.
What do you want to do? Explore all car files or a particular one?
If you want to view module file in demo you need find it's ID and enter it in the Other options input e.g:
-e 10779,11069
IDs can be found in TCOM file (for CLIO IV it's TCOM_155.xml) in the PyRen_home_dir/Vehicle/.
If you want to load all files of a car you need to update pyren and then instead of module IDs you put TCOM file number e.g:
-e 155

AndyIreland, 2020-06-08:
> It's normal, because you have selected demo mode.
> What do you want to do? Explore all car files or a particular one?
> If you want to view module file in demo you need find it's ID and enter it in the Other options input e.g:
> -e 10779,11069
> IDs can be found in TCOM file (for CLIO IV it's TCOM_155.xml) in the PyRen_home_dir/Vehicle/.
> If you want to load all files of a car you need to update PyRen and then, instead of module IDs, you put TCOM file number e.g:
> -e 155

Thank you for the reply, I shall try all of that. No, its just that I downloaded the 09p file and didnt have the car around at that time and wanted to look at what was new by pressing DEMO. 

See if I press DEMO when ' ./pyren ' is in the version box it started up it goes into demo mode no problem , but when I choose 09p in the version box it just exited out of pyren. 

later on today I shall have the car to test on anyway so I wont be using the demo

Marianpol, 2020-06-08:
> Thank you for the reply, I shall try all of that. No, its just that I downloaded the 09p file and didnt have the car around at that time and wanted to look at what was new by pressing DEMO.
>
> See if [...]

Now I probably understand you. You have already used Pyren before with './pyren' version and now you have downloaded 'new' version './pyren09p' (it's quite old actually) and you cannot explore your previous scanned modules in the demo mode?

If so, just go to pyren_home_dir/pyren/, search for savedEcus.p file and then copy it to pyren_home_dir/pyren09p/.
Now in demo mode you should be able to explore your previous scanned modules.

AndyIreland, 2020-06-09:
thank you for your help @Marianpol  - so pyren09p is old now? - can you tell me what latest version  is please?

gobo-ws, 2020-06-10:
@AndyIreland The pyren project can be found at https://gitlab.com/py_ren/pyren

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836 nada temos a ver com o pyren rapaz ...

closed 2023-06-02 by Furtif