cedricp/ddt4all #403: Eco drive setting on rlink 1 -  Megane 3

Ghighi246, 2020-06-16:
Hello everyone,

I have a megane 3 where I replaced (with some external help) carminat tomtom with new rlink navigation. When I installed them the guy who helped said that the eco function will not work because of the cluster/tdb type (and he was right, on eco drive screen all the fields were empty) . Few days ago I replaced the old cluster with a newer model and now I receive on my navigation display in the eco screen the trip computer information but the part with the accelerating, gear change and anticipation does nothing. For the moment the acceleration is 100%, gear change is 0% and anticipation is 100% which results in the eco score on 80/100, but these values won't change now matter how I drive the car.
Can someone help me with some printsceens from their settings? Do I need to fill the eco profile in ddt4all? 
Navigation is composed from MFD v5. 4 and BIC A2/A3 v24. I checker the BIC and it is has the setting to converse on script 1 type the informations. 

Do you have any other ideas why this eco drive screem is not fully functional?

Ghighi246, 2020-06-19:
Dear all,

Today I managed to take some screenshots from the car.

From what I saw and I think that has some connection to the ecodrive function I can post the following:
1) Cluster 
- Cluster type identification from ecu database is CLUSTER_B95_L38_ph2_V1.2 (with the mention not perfect match)
- vehicle type selected is "L". Is this correctly? I check this setting of the cluster on another megane 3 (unfortunatelly without rlink) and it was the same setting.
- I tried to open the ecu CLUSTER_B95_L38_ph3_V1.92 and try to send the following setting to the cluster installed in the car: "Presence MFD - MPU dans MFD". The setting was sended using the button next to the selection, I didn't press the button "Prise en compte de configuration". Should I do that? After selecting this "Presence MFD - MPU dans MFD" i reset navi by pressing 5 times the home button. After this reset no calculation at all: the score is displayed "--/100" and 0 on acceleration/gear change/anticipation. Does anybody knows why?
 
2) MFD - Configuration 
I checked the vehicle type declare in the MFD and it was K95 (with an upper 2 note). Should I select a different type? Maybe one with L letter? As I said before now the score is "--/100" and the bar graphs are empty, but rest of the display is working fine (distance, consumption, etc.)

3) BIC - Configuration
I checked again and this one is set vehicle environment "Script 1 - x38, B32, x91, x95, etc.". But I also checked the serial number of the device and when I searched online this BIC I found that this one was usually installed on clio iv/captur cars. Could this be a problem? Should I make a hard reset of the BIC?

Regardless of the ecodrive function: which type of reset is recommended? Hard reset or soft reset?

Thank you in advance!

Lampetia, 2022-03-27:
Hello. were you able to fix the problem?

kamixae, 2022-05-25:
I have the same problem. I have everything set up, but it doesn't show the eco rating. I heard that the ECU may not send information to the BIC, is that correct? I have Megane III ph2. I would like to add that everything else works properly: sensor visualization, air quality and fuel consumption data. What may be the problem? Could the BIC module be a problem?

AndyIreland, 2022-05-25:
I have a 2013 Renault Zoe electric vehicle fitted with a r-link1 and when i do a search the BIC ECU does not show up . So I am predicting that I have not even got a BIC module in my Zoe ... but however the eco does show up as intended .

kamixae, 2022-05-25:
It may not show up on the list when you are working on a regular CAN bus. You must switch to CAN Multimedia.

AndyIreland, 2022-05-25:
> It may not show up on the list when you are working on a regular CAN bus. You must switch to CAN Multimedia.

yes and the specially soldered ELM327 or specially wired  cable to read the multimedia on CAN

kamixae, 2022-05-25:
> surely you have enabled eco features in the MFD settings: [image]

Yes, the ECO function is on. When I get home, I will send my configuration. Maybe other R-LINK settings are to blame?

AndyIreland, 2022-05-25:
> I doubt you don't have a BIC module. Does R-Link switch on when the door is opened? Are parking sensors displayed and fuel consumption data on the screen?

when i do a search with the magnifying glass in DDT4ALL the BIC module is not one of the found ECU's and if I manually add the BIC from the list then go to read the ECU setting I just got 'NO DATA' in red , so I was presuming there is no BIC module in the Zoe. I could be wrong. 
in Zoe Visu 5 wiring diagrams I cannot find a BIC listed in the list of ECU's also

kamixae, 2022-05-25:
ZOE is not actually listed in the BIC scripts:

Script 1 — x38/x43/x47/B32/x91/x95
Laguna 3, Latitude, Megane/Scenic 3

Script 2 — P32M/H45
Koleos

script 3 J84
Megane/Scenic 2

script 4 J81
Espace

script 5 X83
Trafic

script 6 X61 or X62
Kangoo or Master

script 8 X85
Clio 3

script 9 X44
Twingo 2

script 11 x95RS
Megane 3 RS

script 12 x85RS or x44RS
Clio 3 RS or Twingo RS

script 13 L38VE
Fluence

Script 14 — x07 R*S/x52/x82/x87/x98
Twingo 3, Traffic 3, Logan 2, [Sandero 2](https://www.drive2.com/parts?query=SANDERO2), Duster, Kaptur, Arkana

Script 15 — x61VE
Kangoo 2

kamixae, 2022-05-25:
Apparently you don't.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif