cedricp/ddt4all #404: Not working on Clio V

jaggyjags, 2020-06-18:
I can connect, read all the values but I cannot change any of them, i've heard it is also the case for Captur 2, both using Easy link systems. Is there a work in progress? Anything I can try? Thanks!

KarelSvo, 2020-06-25:
https://www.bosch-mobility-solutions.com/en/products-and-services/passenger-cars-and-light-commercial-vehicles/connectivity-solutions/central-gateway/
The CGW contains a write lock that can only be deactivated via an online connection to the Renault server. You can only try to connect to the ecu in front of the CGW.

EXcomZkko, 2020-07-01:
Same here. I hope someone do something about this security(?) system.

jjaemny, 2020-10-04:
According to a Renault employee, the recently released cars have the gateway locked. I tried coding through ddt4all but stopped writing step. Has anyone solved it?

pilotkalem, 2020-11-04:
You need to connect can lines directly from BCM, not OBD to avoid CGW.

EXcomZkko, 2020-11-04:
> You need to connect can lines directly from BCM, not OBD to avoid CGW.

Any walkthrough about it?

cedricp, 2020-11-04:
> You need to connect can lines directly from BCM, not OBD to avoid CGW.

Was wondering if it was possible, is it working ? If so, you just have to connect CANH/CANL lines from OBD adapter directly to CANH/CANL from ECU, that will bypass the gateway. You may need ECU pinouts, though

pilotkalem, 2020-11-04:
> > You need to connect can lines directly from BCM, not OBD to avoid CGW.
> 
> Was wondering if it was possible, is it working ? If so, you just have to connect CANH/CANL lines from OBD adapter directly to CANH/CANL from ECU, that will bypass the gateway. You may need ECU pinouts, though

When you try to WRITE data to ECU, CGW makes and echo with a part of DDT4ALL write command. So the real command newer arrives to ECU correctly. If you connect CAN Lines directly to BCM then CGW will not work - because you are at internal network. and YES, I TRIED- WORKED. I tried because I sniffed the traffic and saw what CGW do and found this solution. BCM is located under left - down of steering wheel. I dont rebember the pin out but its easy to locate.

EXcomZkko, 2020-11-05:
@pilotkalem So, If we connect to BCM; we can send write command to other ECU's on same the network too?

pilotkalem, 2020-11-05:
> @pilotkalem So, If we connect to BCM; we can send write command to other ECU's on same the network too?

Yes, with that connection you will join internal can network so you can communicate with all other ecu's as well. I said BCM can because it the easiest location to join internal can network.

passyk, 2020-12-05:
Does it mean that ddt4all will have write option Clio V someday?

Yolgezer, 2021-01-01:
Friends Clio V is there any new development regarding the problem of connecting to the vehicle?
Currently, there are those who have achieved this and uploaded the video of the vehicles they have implemented on Youtube.
So it is possible to connect to this vehicle.

KarelSvo, 2021-01-01:
> Friends Clio V is there any new development regarding the problem of connecting to the vehicle?

There are no problems with the connection, you just have to connect the Elm directly to the body control module.

passyk, 2021-01-02:
> > Friends Clio V is there any new development regarding the problem of connecting to the vehicle?
> 
> There are no problems with the connection, you just have to connect the Elm directly to the body control module.

That's true however it's not possible to save changes. Is this still true? Or maybe I have old informations and writing is already implemented?

pilotkalem, 2021-01-02:
Guys, as I said before, you can WRITE parameters to Clio 5 with proper
connection - directly to bcm can lines. I tested many times.

passyk, 2021-01-02:
Is there any manual how to do it?
What kind of interface is needed?

EXcomZkko, 2021-01-02:
Actually we only need BCM pinout for this. Could anyone reach this information?

pilotkalem, 2021-01-02:
Elm327. Just separate can lines from İnternete abd connect direcrly to bcm
thats all.

2 Oca 2021 Cmt 15:13 tarihinde EXcomZkko <notifications@github.com> şunu
yazdı:

KarelSvo, 2021-01-02:
> Actually we only need BCM pinout for this. Could anyone reach this information?

Give a mail address.

EXcomZkko, 2021-01-02:
> > Actually we only need BCM pinout for this. Could anyone reach this information?
> 
> Give a mail address.

excom_zkko@hotmail.com. (Thanks)

KarelSvo, 2021-01-02:
You can remove your address

karadas87, 2021-01-04:
> > Aslında bunun için sadece BCM pinout'una ihtiyacımız var. Bu bilgiye kimse ulaşabilir mi?
> 
> Bir mail adresi verin.

karadas87@gmail.com thanks

karadas87, 2021-01-04:
> > Actually we only need BCM pinout for this. Could anyone reach this information?
> 
> Give a mail address.

karadas87@gmail.com Thanks.

jjaemny, 2021-01-05:
> Guys, as I said before, you can WRITE parameters to Clio 5 with proper connection - directly to bcm can lines. I tested many times. 2 Oca 2021 Cmt 04:05 tarihinde passyk <notifications@github.com> şunu yazdı:
> […](#)

Can you show me the picture of connecting BCM and ELM327? I am a Korean who is riding the xm3 called Arkana in a foreign country. Arkana can be coded using elm327, but xm3 cannot be stored as modified values. I ask for your help me. Thank you.

jordirs2004, 2021-01-05:
> > Actually we only need BCM pinout for this. Could anyone reach this information?
> 
> Give a mail address.

first, thx for give me this topic.

Is it possible to have this information please? jordivpn.31@gmail.com

thx

giammi553, 2021-01-05:
> > Actually we only need BCM pinout for this. Could anyone reach this information?
> 
> Give a mail address.

Can you send me that information ? Thank you very much in advance!
void2056@gmail.com

jordirs2004, 2021-01-08:
hi all,
is one you tried to connect directly on the BCM? does-it work?
Thx

KarelSvo, 2021-01-11:
> When you try to WRITE data to ECU, CGW makes an echo with a part of DDT4ALL write command. So the real command newer arrives to ECU correctly. If you connect CAN Lines directly to BCM then CGW will not work - because you are at internal network. and YES, I TRIED- WORKED. I tried because I sniffed the traffic and saw what CGW do and found this solution. BCM is located under left - down of steering wheel. I dont rebember the pin out but its easy to locate.

Have you only observed the free exchange of data between the control units?
Or have you reached all control devices from the BCM via ISOTP Can address? ISOTP access is required to change configurations. Research so far has shown that there are various gateway channels to which several control units are assigned.
If you connect directly to the BCM, only the control devices of the channel to which the BCM is assigned can be reached via ISOTP.
Radio (multimedia channel 2) cannot be reached from the BCM (general channel) via ISOTP.
Did you have other experiences?

gacocz, 2021-01-23:
> > Actually we only need BCM pinout for this. Could anyone reach this information?
> 
> Give a mail address.

Covid you kindly send it to me too? 
gaetano.corrini@gmail.com

Probably are the white/green wire but any help will be needed. 
Thanks

ustunkaya07, 2021-01-31:
> > Actually we only need BCM pinout for this. Could anyone reach this information?
> 
> Give a mail address.

> > Actually we only need BCM pinout for this. Could anyone reach this information?
> 
> Give a mail address.
could you please send to 
kayaustun@hotmail.com
thanks,

pboleta, 2021-02-26:
> > Actually we only need BCM pinout for this. Could anyone reach this information?
> 
> Give a mail address.

Can you send me that information ?
ptboleta@hotmail.com
Thanks!

gacocz, 2021-02-26:
Sorry, I don't have any wiring diagram or pinout to share.

Anyway if you want I can send you my progress.

I want to cut the Can L and Can H pin on my OBD interface and put it
directly in other can network point, when I will found the correct point
maybr i can write because sgw is bypassed...

pboleta, 2021-02-26:
Hi,
I appreciate if you share your information and progress, I would be very grateful.
I once spoke to a German guy who provided me with this information, but which I have not tested as I still have my car under warranty. I don't know if it is useful to you or you were already aware of it. Anyway, here is the attachment.

Regards
Pedro

gacocz, 2021-02-26:
I don't see the attachment...

pboleta, 2021-02-26:
I will send it again https://1drv.ms/w/s!Agvr4ZvRqCbqgbsxeffU0T0YB1uHdQ?e=TNEXqs

gacocz, 2021-02-26:
Thanks,

That's what i want to do, I will make some test and keep you updated 😉

eminpolat85, 2021-02-27:
> > Actually we only need BCM pinout for this. Could anyone reach this information?
> 
> Give a mail address.

ozgun@live.de thanks

mnigten, 2021-02-27:
Can you send it to me as well? Nigten@gmail.com 

Thanks

eminpolat85, 2021-02-27:
hasn't come yet

iOS için Outlook<https://aka.ms/o0ukef> uygulamasını edinin
________________________________
Gönderen: Michael Nigten <notifications@github.com>
Gönderildi: Saturday, February 27, 2021 11:03:23 AM
Kime: cedricp/ddt4all <ddt4all@noreply.github.com>
Bilgi: eminpolat85 <ozgun@live.de>; Comment <comment@noreply.github.com>
Konu: Re: [cedricp/ddt4all] Not working on Clio V (#404)

Can you send it to me as well? Nigten@gmail.com<mailto:Nigten@gmail.com>

Thanks

KarelSvo, 2021-02-27:
> I will send it again 

You can delete these findings because the original Renault documentation does not match the real BCM Ecu.
Renault assigned connections incorrectly in the documentation.

pboleta, 2021-02-27:
Really? You really can't believe anything, not even the documentation of a brand.
I apologize for giving wrong information.
Any developments please let me know.

gacocz, 2021-02-27:
Pinout it isn't a problem. 
Renault uses green and white for CAN Network wiring and also you can measure tension referring to negative and the resistance between them. 

So those are a simple thing we can check before start trying to use OBD interface and DDT4ALL.

KarelSvo, 2021-02-27:
> Renault uses green and white for CAN Network wiring and also you can measure tension referring to negative and the resistance between them.

On my Easylink radio ecu, CAN cable are white, purple, and salmon-colored.

gacocz, 2021-02-27:
> > Renault uses green and white for CAN Network wiring and also you can measure tension referring to negative and the resistance between them.
> 
> On my Easylink radio ecu, CAN cable are white, purple, and salmon-colored.

Why 3 cables? 
How did you found it? Did you measure it?
Did you try to put pin 6 and 14 of the OBD interface 
 directly there?

KarelSvo, 2021-02-27:
> Why 3 cables?

No 3 different colors. 
I just want to say that the cable color is not a safe criterion.
This radio has 6 connections for CAN cables.

jordirs2004, 2021-02-27:
[image]
[image]
[image]
[image]
[image]
[image]

gacocz, 2021-02-27:
> > Why 3 cables?
> 
> No, 3 different colors.
> I just want to say that the cable color is not a safe criterion.
> This radio has 6 connections for CAN cables.
> You cannot reach the radio from the BCM. It is assigned to a different gateway channel.

I supposed so, maybe radio connects directly to the SGW, the SGW should be located under the cluster, where did you found the radio ECU? 

If I connect directly to CAN H and Can L of the RADIO ecu do you think I will able to change HMI settings? 

I would like to change display and I think I should change display format ...

gacocz, 2021-02-27:
So, according to this Bosch scheme, there should be 4 channels for the Can Network behind the SGW, if we locate (under the cluster?) We could connect can h and can l in according to the domain we need.

I suppose that if I connect to the channel of the multimedia domain and scan the ECU I should found just the one related to it...

cedricp, 2021-02-27:
You can also search for twisted pairs. Thats how I generally do

KarelSvo, 2021-02-27:
> So, according to this Bosch scheme, there should be 4 channels for the Can Network behind the SGW, if we locate (under the cluster?) We could connect can h and can l in according to the domain we need.

jordirs2004 has already done a lot of research. But unfortunately no success when connecting directly to the radio.
Maybe he'll show photos of it too.
On the BCM, the wires are not twisted. There he found all the control units for the motorization and the vehicle channel.

gacocz, 2021-02-27:
> jordirs2004 has already done a lot of research. But unfortunately no success when connecting directly to the radio.
> Maybe he'll show photos of it too.
> On the BCM, the wires are not twisted. There he [...]

So even if you connect directly to the Can Network behind the SGW you can't write ecu? 

Very strange...

jordirs2004, 2021-02-27:
[image]

gacocz, 2021-02-27:
@jordirs2004 that's exactly what I want to do.
If this didn't work what kind of error did you had? Did you logged some debug?

jordirs2004, 2021-02-27:
I have nothing, no datas in ddt4all
I don't have the log files with me but Karelsvo saw it

gacocz, 2021-02-27:
> I have nothing, no datas in ddt4all
> I don't have the log files with me but Karelsvo saw it

Did you try to connect on other ECU, for example on the BCM?  Just to know if this problem is related only to the multimedia subsystem or not.

Maybe ecu need authenticated messages by SGW that forward read messages, but block write messages so if you bypass it your read messages will be ignored by ecu.

If it's work in this way it will be hard to write ecu without Renault server 😕

gacocz, 2021-03-12:
I've made a lot of test but unfortunately, like jordis2004 says, I receive NO DATA when connect directly on the internal Can Network. 
Those who say they are able to read data from internal network could kindly explain in detail how do they did?

I connected pin 6 and 14 directly on the BCM (100% sure can network, I've checked with multimeter) but NO DATA.

It's seems to me that internal Can network isn't open lime we suppose...

eminpolat85, 2021-03-12:
Can anyone solve ozgun@live.de

cetinyusuf, 2021-03-15:
sonuç olarak bağlanabilen oldumu clio 5 @jordirs2004  @KarelSvo  bağlanabilen olduysa ben nasıl yapabilirim

ulasatar, 2021-03-21:
Çözen varsa bana da bilgi verebilir mi? Can anyone give me information?

cetinyusuf, 2021-03-21:
@ulasatar  türkiyede bir kişi yapabılıyor oda PROCAR CAMERA bir abim yapıyor tabi dünyada şuana kadar SGW geçebilen bir kişi var oda o

alpi1, 2021-03-28:
Ben kendi aracım için yaptım

[509 more replies]

closed 2023-06-02 by Furtif