cedricp/ddt4all #408: Renault Megane 4 "diag has been set invalid value gen1_1 " Error

Flavius08, 2020-06-21:
Hi group I also have a problem if someone can help me I would be grateful.  I entered Developer mode and changed 2 settings (first picture) to both it wrote disabled and I set as in the first picture, SPORT and COMFORT and after I saved and restarted the radio I got an error (picture 2) and  I can't do anything anymore, please if anyone can help me.
[image]
[image]

Joakim02, 2021-01-14:
Hello, did you solve your problem?

Flavius08, 2021-01-14:
Yes, I reset the system with a diagnostic device. I don't know exactly how I did it with my uncle.I had 3 diagnostic devices but only one worked

Joakim02, 2021-01-14:
OK thanks you.

sallythe, 2021-10-05:
> Yes, I reset the system with a diagnostic device. I don't know exactly how I did it with my uncle.I had 3 diagnostic devices but only one worked

hi Flavious08 am having the same trouble as you can you tell how to remove it. thanks

sallythe, 2021-10-05:
hi Flavious08 am having the same trouble as you can you tell how to remove it. thanks

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

firatkacmaz72, 2023-06-20:
I HAVE THİS PROBLEM TOO PLEASE HELP ME

gontsendala8-rgb, 2025-08-26:
Hi anyone who was able to reset

karim03-afk, 2025-09-05:
[image]
Bonjour à tous j’ai ce problème là pouvez vous m’aider s’il vous plait j’ai tout essayer (débrancher la batterie, les fusibles éteindre l’écran ect..)

thamid375, 2026-11-01:
> 10\. [image]
>      Bonjour à tous j’ai ce problème là pouvez vous m’aider s’il vous plait j’ai tout essayer (débrancher la batterie, les fusibles éteindre l’écran ect..)
ta trouver une solution ?