cedricp/ddt4all #409: ddt4all doesn't work Ubuntu/Xubuntu 'focal'

robertalks, 2020-06-25:
Hello, I upgraded my machine to Xubuntu 'focal' 20.04, but I can't run ddt4all at all anymore. python2 doesn't have qt4 libs, they have been removed and with python3 ddt4all doesn't work due to the code:

`robert@pandora:~/work/git/ddt4all$ python3 ./ddt4all.py 
  File "./ddt4all.py", line 260
    print name, group
          ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(name, group)?
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 72, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
  File "/usr/lib/python3/dist-packages/apport/__init__.py", line 5, in <module>
    from apport.report import Report
  File "/usr/lib/python3/dist-packages/apport/report.py", line 13, in <module>
    import fnmatch, glob, traceback, errno, sys, atexit, locale, imp
  File "/usr/lib/python3.8/imp.py", line 19, in <module>
    from importlib._bootstrap import _ERR_MSG, _exec, _load, _builtin_from_name
ModuleNotFoundError: No module named 'importlib._bootstrap'

Original exception was:
  File "./ddt4all.py", line 260
    print name, group
          ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(name, group)?
`

cedricp, 2020-06-25:
Try with **python3** branch : https://github.com/cedricp/ddt4all/tree/python3

robertalks, 2020-06-25:
Now I get a different error:

`Traceback (most recent call last):
  File "./ddt4all.py", line 35, in <module>
    import parameters, ecu
  File "/home/robert/work/git/ddt4all-python3/parameters.py", line 4, in <module>
    import ecu, elm
  File "/home/robert/work/git/ddt4all-python3/ecu.py", line 5, in <module>
    import zipfile
  File "/usr/lib/python3.8/zipfile.py", line 8, in <module>
    import importlib.util
ModuleNotFoundError: No module named 'importlib.util'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 72, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
  File "/usr/lib/python3/dist-packages/apport/__init__.py", line 5, in <module>
    from apport.report import Report
  File "/usr/lib/python3/dist-packages/apport/report.py", line 13, in <module>
    import fnmatch, glob, traceback, errno, sys, atexit, locale, imp
  File "/usr/lib/python3.8/imp.py", line 19, in <module>
    from importlib._bootstrap import _ERR_MSG, _exec, _load, _builtin_from_name
ModuleNotFoundError: No module named 'importlib._bootstrap'

Original exception was:
Traceback (most recent call last):
  File "./ddt4all.py", line 35, in <module>
    import parameters, ecu
  File "/home/robert/work/git/ddt4all-python3/parameters.py", line 4, in <module>
    import ecu, elm
  File "/home/robert/work/git/ddt4all-python3/ecu.py", line 5, in <module>
    import zipfile
  File "/usr/lib/python3.8/zipfile.py", line 8, in <module>
    import importlib.util
ModuleNotFoundError: No module named 'importlib.util'
`

cedricp, 2020-06-26:
I work with Ubuntu/focal and it's working fine for me

robertalks, 2020-06-26:
Well, thats all fine and dandy, but whats the error from? I might be missing some module, but I don't know which.

robertalks, 2020-07-04:
The problem is 'importlib/' which is available in ddt4all, once removed, it works.

> I work with Ubuntu/focal and it's working fine for me

mecamaniack44, 2020-12-26:
bonjour.
Je viens de télécharger le dossier de cedricp sur github "python3.py"
Pouvez-vous me dire comment l’installer sur Ubuntu 20.04.
merci d'avance.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif