cedricp/ddt4all #410: After enabling ACC, I'm getting Active Breaking disconnected.

robertalks, 2020-06-26:
Hello, I tried to enable ACC using ddt4all. I got it enabled and after and now I'm getting that the Active Breaking is disconnected. Any idea how to enable it or what I did wrong?

[image]

Any idea would really help.

scoobydoo511, 2020-06-28:
Check your AeBs configuration in radnav section. Maybe you have deactivated
by mistake....

pet, 26. lip 2020. 22:31 Robert Milasan <notifications@github.com> je
napisao:

robertalks, 2020-07-01:
> […](#)

Doesnt work, any other ideas?

mmitech, 2021-05-15:
@robertalks any luck with getting the ACC working? if so could you please let me know how you did it?

robertalks, 2021-05-16:
No, I didnt get it to work, nobody seems to know the exact 'settings' to make it work.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif