cedricp/ddt4all #411: Changing WAC Jingle

AndyIreland, 2020-06-30:
hello all - anyone got any ideas how to change the WAC (walk away config) Jingle/bleep? - i can disable it in another screen but I dont want to do that I just want to change the Hz so it will change the tune when the car locks when I walk away and the car locks. I have looked in the manual command in DDT4ALL and it cannot be changed in that either. Thanks.
[image]

JABICHOCU, 2020-07-20:
Same story here on my 2013my Captur, I have the option of turning it off but nothing like changing sound level as in the parking system buzzer

JABICHOCU, 2020-07-20:
I don´t see any way of changing this 
[image]

JABICHOCU, 2022-10-09:
Hello again ,AndyIreland...any advances on this one?  As I said before, there´s no way on my car to change those parameters. On the configuration section of the HFM module I have this options...do you know what they do? regards
[image]

AndyIreland, 2022-10-09:
> Hello again ,AndyIreland...any advances on this one? As I said before, there´s no way on my car to change those parameters. On the configuration section of the HFM module I have this options...do you know what they do? regards [image]

no further forward in changing the parameters - only think they can be read . I am not sure if you can manually change using one of these buttons in DDT4ALL on the top bar:

[image]

WAC - is walk away config - if you were to set that to 'false' then the 'blip blip' sound will be disabled as you walk away from the car.

VSC stands for 'keyless vehicle' 

not sure what the other settings do so I didnt even attempt to change them just in case an error.

I dont think any of them changed the actual sound of the 'blip blip' tune though

KarelSvo, 2022-10-09:
If the sound element is a bugle like the Duster II, it cannot produce a melody.

JABICHOCU, 2022-10-09:
It´s a pity...besides being too loud, I think it is too long as well. Thanx anyway

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif