cedricp/ddt4all #412: CLIO IV - SubFunction Not Supported

InvalidIP, 2020-07-09:
Hello,

 I have a CLIO IV 1.2 55kw (2015) and I have downloaded the latest version (20.07.2019) from the Russian site. I also have a BT adaptor with the PIC18F25K80. I want to disable TPMS by using X98 Clio IV > Cluster X98_X87_X82_xA_x98ph2_xHAph2_v7.8 (it automatically detects this when scanning) > Ecran Config TdB > Config Generale2 and the Param Tpms_Present => "Without TPMS" right now it is set "With Direct TPMS". Everything seems to work fine (ELM adapter working, sending data indicated by green light), once I apply the setting the left tab (LECTURE) also shows the new setting but I can see in the log: 

19:30:47.975;745;1003;7F1012;NR: SubFunction Not Supported
19:30:48.284;745;2180;61 80 35 36 33 39 52 04 30 30 31 30 30 30 30 30 14 19 0B 75 0B 75 02 01 00 88 
19:30:48.744;74D;1003;7F1012;NR: SubFunction Not Supported

And after restarting the dash, the car doesn't show the TPMS error but the STOP light remains ON and and the dash is showing ERROR PROG. I've searched russian forums, italian forums, turkish forums but can't get to the bottom of it. 

 Anyone has any ideas? tried also PyREN but really not sure how to use it, as it seems to need decimal inputs.

pakoto, 2020-07-09:
ERROR PROG dissapear if you revert to without tpms. I´m searching for a solution like you. 

Maybe need to modify options in abs ecu before tdb?

InvalidIP, 2020-07-09:
Hey, that was fast :) Yeah, if I revert everything goes back to normal, but still I hate that TPMS error in the dash. 
Also, the Fron_Sbr Inhibitor appears as activated when I read the ECU but it still beeps like hell. I don't understand what's the issue.

EXcomZkko, 2020-07-11:
You have to disable tpms on bcm module too.

InvalidIP, 2020-07-14:
Thank you so much, issue resolved.

closed 2020-07-14 by InvalidIP

pakoto, 2020-07-14:
Where is tpms option in bcm?

InvalidIP, 2020-07-14:
> Where is tpms option in bcm?

[BCM/UCH] > TPMS System > TPMS Configuration and modify TPMS_Presence_SSPP from 1 to 0, after this everything should work fine.