cedricp/ddt4all #413: Start-stop inhibition

JABICHOCU, 2020-07-11:
Hello to all of you, I´ve seen that on many cars it ´s possible to access "STOPSTARTINHIBIT" on the BCM  by changing configuration on "GEE" options...but I don´t have this on my BCM :-1: 
[image]

On my 2013 Captur , I have found  a similar thing on  T4_BCM_DDT2000_SW11_0_20130128T102714  with these  options in the "alternator management" section

[image]

Will this be the way to inhibit S&S ??? I have fear of bricking the module by doing this...any help would be much appreciated, thanks :+1:

JABICHOCU, 2020-07-16:
No ideas?? @Pakoto...where are you??? :)

pakoto, 2020-07-16:
EMM = lights controller
What is your BCM? t4....

JABICHOCU, 2020-07-21:
Hello Pakoto...good to see you around here. This is my BCM >
[image]

But this options I see them on the EMM module

KarelSvo, 2020-08-24:
Better to deactivate the stop start in the engine control unit.

JABICHOCU, 2023-04-04:
> Better to deactivate the stop start in the engine control unit.

Hello Karelsvo, can you tell me which module is the control unit on Captur x87?

AndyIreland, 2023-04-04:
if you have not got it listed in your ECU then what can you do apart from changing the BCM module to a later one or try and update the SW11 to a later software number (if that is even possible) 
Thats why I said in your other post , why not remove the stop & start lead off the 12v battery under the bonnet then extinguish the warning light and message on the dash about stop start ?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif