cedricp/ddt4all #414: R-link 2 Boot-loop after upgrading from V2 to V7

estyasser, 2020-07-11:
Hello everyone,

Today I tried to upgrade my R-link 2 software from V2.2.19.300 to V7.0.24.160 which I found over the internet. The update started normally and after 4 reboots, it showed remaining time 4m and rebooted then it got stuck on boot loop and the ambient lights on the instrument panel (cluster) is flashing, I left it like that for more than 30 minutes with the engine on and door closed.
I tried soft reset and hard reset from ddt4all and from the on/off button on the screen and still not working. I also tried to turn off the car and start it again but no result.

I have 2017 Renault Talisman with 8.7” touchscreen.

Is there anyway to fix this or downgrade to 2.2.19.300? Or can I flash the v2 firmware using ddt4all?

Attached re the log files extracted from the USB

Thanks all :)

[Upgrade0.log](https://github.com/cedricp/ddt4all/files/4907695/Upgrade0.log)
[Upgrade1.log](https://github.com/cedricp/ddt4all/files/4907696/Upgrade1.log)

estyasser, 2020-07-12:
Any help please?

tomidix, 2020-07-12:
It is not possible to upgrade v2 (nor v3) by yourself.
Upgrade is done exclusively by your dealership and it is by no means an ordinary upgrade process.
It is basically a reprogramming process which is done if customer's RLink has some kind of a problem. They do a reprogramming of the device and during the process bootloader and software are both updated. This is more like a fix for problematic devices than some kind of update.
I'm afraid that you have flashed a firmware which requires newer bootloader than you have so your device won't boot..
You can try this but I just think you will have to go to your workshop now:
https://gitlab.com/py_ren/pyren/-/wikis/how%20to%20rlink2%20black%20screen

estyasser, 2020-07-12:
> It is not possible to upgrade v2 (nor v3) by yourself.
> Upgrade is done exclusively by your dealership and it is by no means an ordinary upgrade process.
> It is basically a reprogramming process which [...]

Thanks for your reply. Unfortunately it didn't work. I tried it but the method mentioned above is for screen parameters, when the system is running but the screen is black. The problem I have is the bootloop, where the system never runs and keep rebooting over and over. 

If there's a way I can force flashing the 2.2.19.300 version from ddt4all it's be great. maybe using python or something if it's not possible with ddt4all.

Thanks.

tomidix, 2020-07-13:
As you said, you have stucked in a bootloop aka bricked your device.. If the device won't boot, you obviously won't be able to flash it ordinary way.
Flashing Rlink2 device from service modes is not achievable by ddt4all nor any other python or xyz tool at the moment.
Actually, on numerous forums there are warnings that v2 to v7 is done exclusively by workshops. I don't know how and why you have tried to do this on your own?

Anyway, I hope you're gonna be able to get your device back at your workshop and won't have to buy a new one.

estyasser, 2020-09-26:
@tomidix @shrlnm From your experience guys, can I reprogram RLink2 from using Can Clip? If yes then how?

AlMeu, 2021-02-12:
I have an unit v2.2 in which someone try upgrade it to v3.3.
Now is starting and get first warning message and when is getting first screen is restarting itself.
There is any chance to put in a stable state, to reflash it?

BillyJacobie, 2021-04-16:
I have been having this issue since 2019 when I updated my Rlink from 2.2.19.300 to v8
I must admit that I realize only now that it was my mistake because I did not do enough research and only found out afterwards that my HARDWARE does not support V8, it never came on again!

Today I finally managed to restore my unit and brought it back to life. 
After 2 years of struggling, I think it might have been lucky but I believe I found a way to resolve the issue of the boot loop for units that have a green flash on the **Control Panel** that is stuck in the boot loop after upgrading from V2 to V7 or V8.

I will compile a detailed step by step for and make available to this issue.

BillyJacobie, 2021-04-16:
Hi,
I think this could be related?
https://github.com/cedricp/ddt4all/issues/430#issuecomment-821377409

AlMeu, 2021-08-01:
I test and didn't work.

dimdarzov, 2023-03-26:
> Привет всем,
> 
> Сегодня я попытался обновить программное обеспечение R-link 2 с V2.2.19.300 до V7.0.24.160, которое я нашел в Интернете. Обновление началось нормально, и после 4 перезагрузок оно показало оставшееся время 4 м и перезагрузилось, затем застряло на петле загрузки, и мигает окружающее освещение на панели приборов ( кластер ), Я оставил это более 30 минут с включенным двигателем и закрытой дверью. Я попробовал мягкий сброс и жесткий сброс с ddt4all и с кнопки включения / выключения на экране и все еще не работал. Я также попытался выключить машину и запустить ее снова, но безрезультатно.
> 
> У меня есть Renault Talisman 2017 года с сенсорным экраном 8,7 ”.
> 
> Есть ли что-нибудь, чтобы исправить это или понизить до 2.2.19.300? Или я могу запустить прошивку v2, используя ddt4all?
> 
> Прикреплены файлы журнала, извлеченные из USB
> 
> Спасибо всем: )
> 
> [Upgrade0.log](https://github.com/cedricp/ddt4all/files/4907695/Upgrade0.log) [Upgrade1.log](https://github.com/cedricp/ddt4all/files/4907696/Upgrade1.log)
> 
> #renault #talisman #megane #rlink #r-link #bootloop #boot-loop

> Hello everyone,
> 
> Today I tried to upgrade my R-link 2 software from V2.2.19.300 to V7.0.24.160 which I found over the internet. The update started normally and after 4 reboots, it showed remaining time 4m and rebooted then it got stuck on boot loop and the ambient lights on the instrument panel (cluster) is flashing, I left it like that for more than 30 minutes with the engine on and door closed. I tried soft reset and hard reset from ddt4all and from the on/off button on the screen and still not working. I also tried to turn off the car and start it again but no result.
> 
> I have 2017 Renault Talisman with 8.7” touchscreen.
> 
> Is there anyway to fix this or downgrade to 2.2.19.300? Or can I flash the v2 firmware using ddt4all?
> 
> Attached re the log files extracted from the USB
> 
> Thanks all :)
> 
> [Upgrade0.log](https://github.com/cedricp/ddt4all/files/4907695/Upgrade0.log) [Upgrade1.log](https://github.com/cedricp/ddt4all/files/4907696/Upgrade1.log)
> 
> #renault #talisman #megane #rlink #r-link #bootloop #boot-loop

Hello. Please tell us what happened next with the R-
link? I have this problem and don't know how to solve it.

AlMeu, 2023-03-26:
@dimdarzov if it is as described there, your unit it is bricked. Need to change with other good. 
I can help with another units old or new. If you want, leave an email for contact you.

sliskm, 2023-05-30:
Hello everyone, I have same situation. unit bricked, in a boot loop after upgrade not finished. I ordered the same old unit.  Can I count on you @dimdarzov to set it up after installation on car. Some say that it will be locked, that it needs to change vin number, etc... Not sure what exactly needs to be done. Thanks in advance

sliskm, 2023-05-30:
Hello everyone, I have same situation. unit bricked, in a boot loop after upgrade not finished. I ordered the same old unit. Can I count on you @AlMeu to set it up after installation on car. Some say that it will be locked, that it needs to change vin number, etc... Not sure what exactly needs to be done. Thanks in advance

AlMeu, 2023-05-30:
@sliskm you made wrong to order same OLD unit and even locked.
But as you like. Leave an email and I can try help you.

sliskm, 2023-05-30:
> @sliskm you made wrong to order same OLD unit and even locked. But as you like. Leave an email and I can try help you.
 I can still cancel order. my mail is majcinad@gmail.com. thank you

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

Proton9999, 2024-02-08:
Good evening. I downgraded the rlink firmware and then installed the newest one. The menu button is now inactive on the screen. Resets when pressed. Who can help?

Ocsi1980, 2024-03-09:
Sziasztok! Nekem megane 4 GT van és "sos hívás ellenőrizze" írja ki, de nincs benne gyárilag sos hívás. Ezt hogyan lehet javítani ddt4all segítségével?

Ocsi1980, 2024-03-09:
Hello!  I have a megane 4 GT and it says "sos call check", but it doesn't have sos call by default.  How to fix this with ddt4all?

AndyIreland, 2024-03-16:
> Hello! I have a megane 4 GT and it says "sos call check", but it doesn't have sos call by default. How to fix this with ddt4all?

It might be found in your DCIM / TCU parameters - like this:
[image]

Ocsi1980, 2024-03-16:
thanks i will try it

Ocsi1980, 2024-03-16:
I hope I can find it because it's very confusing

AndyIreland, 2024-03-16:
make sure you do a search for your ecu's first using the magnifying glass in ddt4all to make sure it chooses  the right tcu ecu of your car I done TCU gen2 v3.2 as an example you might have a different ECU in yours.
Good luck.

Ocsi1980, 2024-03-17:
i tried but my ecu is tcu_gen2.5_v3.3 i couldn't find the ecall menu in it.  I found what you wrote, but it's not mine

AndyIreland, 2024-03-17:
your right it does not appear to be - what you could do, but it might be dangerous so do at your own peril is you could call up TCU gen2 v3.2 and look at the SOS call settings and if they read and are not in the color red , then try changing the settings from there and see how you get on - but as i say, it could be dangerous, dangerous in the sense that there is always a possibility that it could corrupt your tcu ecu and you might have to get it re-programmed or replaced. 
At the very least doo a backup DUMP of your TCU through PyREN , just in case  you have to try rolling back / restoring the settings

Ocsi1980, 2024-03-22:
Köszönöm válaszod!
A helyzet az, hogy egyszer szűz módba raktam a kijelzőm. Úgy tudtam visszahozni, hogy újra feltöltöttem rá egy szoftvert, de szerintem nem a megfelelőt választottam mert az autóban gyárilag nincsen S.O.S funkció, de világít a check lámpa. Most keresem a gyári Renault Megane GT szoftvert, hogy visszatöltsem azt.