cedricp/ddt4all #415: Opel Vivaro 2013 (like Renault Trafic 2 PH3) - CAN line issue for ELM327 Bluetooth (K line works ok)

djsweat, 2020-07-14:
Hello, 
I have an error connecting the CAN network on the car: Opel Vivaro 2013, similar to Renault Trafic 2, 

I have 2 devices elm327, both interfaces on Bluetooth, with STN1170 chip and with PIC18F25K80 chip, both devices works well for OBD2 but only on the K line modules on my car, so I can see and edit the SRS airbag module and UCH module, tried on Android and PC. For example I could activate 'Follow me Home' option in UCH and I could see the resistance of the airbags in the SRS airbag module, and also other information like keys code etc...

Unfortunately no CAN line modules are available, the CAN line is working because the OBD2 diagnostic is working in the Torque app, ScanTool and other OBD2 app. I tried on PC and Android, 

In DDT4ALL the CAN line generates the same result on searching: CAN error and 0 modules. I tried to increase the response time, 200, 400, 600, 800 and 1000 ms but it doesn't work, Also I open the ecu data base individual without results. I also search the forum for other topics and I didn't find the same error I have,

I think the issue is DDT4ALL can not communicate with my CAN line network and I guess the CAN initialization string is the reason but I do not know how to check, and please if you can help me with an idea what could i do?

The OBD2 connector have next PINs wired: 3,4,5,6,7, 11,12,13,14, 16.

Regards,

Marianpol, 2020-07-14:
I know what the issue is because I faced it myself on my Lagnua 2 PH 2.

Your car have CAN 250 kbps not CAN 500 kbps like most of the cars equipped with CAN.
CAN 250 kbps is set up by ELM command AT SP 8 and CAN 500 kbps is set up by AT SP 6.

In DDT4ALL, CAN protocol baudrate is determined by this line in the module file (CAN 500 kbps):
<CAN BaudRate="500000" Multipoint="1">
CAN 250 kbps:
<CAN BaudRate="250000" Multipoint="1">

But not all files, that are for CAN 250 kbps modules, have this exact line. In some files, like in files for your modules, this line is a little bit different:
<CAN BaudRate="10400" Multipoint="1">

DDT4ALL code checks, if baudrate in this line is 250000, which is not true, so it selects wrong CAN protocol CAN 500 kbps with ELM command AT SP 6.

And thats the problem. DDT4ALL code has no exception for this type of baudrate description.

You can try to use DDT mode in PyRen instead of DDT4ALL.

Marianpol, 2020-07-14:
I dived a little deeper into this problem and I solved it.

In ecu.py change line:
`if self.baudrate == 250000` to `if self.baudrate == 250000 or self.baudrate == 10400:`

And line:
`ecu_conf['brp'] = 1` to `ecu_conf['brp'] = "1"`

djsweat, 2020-07-14:
Thanks for the idea, I tried the changes, but the CAN line didn't work.
Your answer was good for me because I tried to change the CAN speed in the "elm.py" file and I replaced all the values like "AT SP 6" with "AT SP 8", so now ELM works with the speed of 250000, with this change, I managed to see 3 ecu-s in the automatic search and others in the manual selection.
:)
Greetings

KarelSvo, 2020-08-30:
> I dived a little deeper into this problem and I solved it.
> 
> In ecu.py change this line:
> `if self.baudrate == 250000` to `if self.baudrate == 250000 or self.baudrate == 10400:`
> 
> And a line below:
> `ecu_conf['brp'] = 1` to `ecu_conf['brp'] = "1"`

Hi @Marianpol 
In pyren, tcom 155 clio 4 shows MFD ecu is 500kbps brp 0.
The ddt4all MFD file shows <CAN BaudRate = "10400" Multipoint = "1"> 250kbps brp1 and selects SP8.
How does pyren choose in ddt mode?
How can ddt4all select the correct SP6 protocol in this example?

Marianpol, 2020-08-31:
> Hi @Marianpol
> In pyren, tcom 155 clio 4 shows MFD ecu is 500kbps brp 0.
> The ddt4all MFD file shows `CAN BaudRate = "10400" Multipoint = "1"` 250kbps brp1 and selects SP8.
> How does pyren choose in ddt mode?
> How can ddt4all select the correct SP6 protocol in this example?

I didn't know about that. 
So in this example DDT4ALL cannot distinguish if it is CAN250kbps or CAN500kbps module.

Pyren can choose correct protocol, because it is looking into the addresing.xml file of the choosen car that can be found in the DDT2000data/Vehicles folder. 
DDT4ALL uses only DDT2000data/ecus folder, so in this example, it will not choose correct protocol.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif