cedricp/ddt4all #416: Changing buzzer sound level on parking system

JABICHOCU, 2020-07-16:
Hello to you @ll again, I´ve been researcing on how to change buzzer sound on my 2013 Captur cause it´s always been a bit loud and distorted. On my UPA parking system unit (original one by Valeo) I´ve managed to find these options>

 [https://ibb.co/CJzr9fF ](url)

As you can see it says "buzzer sound index=LEVEL 3" I´ve changed this to other levels to see if it works but it doesn´t change the sound...any ideas on this one

ulasatar, 2020-07-23:
it doesn't matter.
Make cfg.BuzzerFrequency.Reardetection====1000hz. 
This will increase the volume a little

JABICHOCU, 2020-07-24:
OK, but it is the opposite.I want to decrease the volume!

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif