cedricp/ddt4all #425: Virgin Mode Renault IV GTline

Emaromao123, 2020-08-01:
Good  Evening,

Im trying to solve my problem with RLINK 2.

I've pressed the "Virgin Mode" Button on the Developer side menu, wrongly.

Now how can i solve this? There is some way to fix without going to Renault?

Maybe with dt4all? I have Black screen, no radio and no lights on Doors inside.

Thank you.

tomidix, 2020-08-10:
You can try this:
https://gitlab.com/py_ren/pyren/-/wikis/how-to-rlink2-black-screen

Emaromao123, 2020-08-10:
> You can try this:
> https://gitlab.com/py_ren/pyren/-/wikis/how-to-rlink2-black-screen

I've tried before and nothing happens... 
I'm using elm 327 USB for now, with Bluetooth I can't connect. When I execute the macro choosing all the different available nothing happens.. Any idea how I can solve this? With Dt4all it seems that can open the app but I can't conect to the elm.

Thank you

maddpower, 2021-10-10:
Have you fixed the problem?
I'm on the same boat...I don't know how to fix this.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

stocky777, 2024-08-30:
> You can try this:
> 
> https://gitlab.com/py_ren/pyren/-/wikis/how-to-rlink2-black-screen

Where are the files?

AndyIreland, 2024-08-30:
I didnt think clio 4 came with an r-link 2 - thought clio 4 came with medianav / medianav evo / rlink1 or medianav2 ?
Are you sure its rlink2 have you a picture of it?