cedricp/ddt4all #430: total black screen on Rlink2.2

motonu1, 2020-08-07:
i was out with my talisman2017 & turned it off after10 min i returned to my car & suddenly i noticed that my console wont work. i touched the power bottom in order to restart it but it want useful. i pulled the plug from the battary under the hood & after a while  i reconnect it .after that also the meter screen got changing the color continuously.
how can i fix it?!!!!

Tony2764, 2020-08-24:
Bonjour, avez vous résolu le problème ?

Cordialement

Anxiro, 2020-08-26:
Did you try this? https://gitlab.com/py_ren/pyren/-/wikis/how-to-rlink2-black-screen
It's the only reference I could find.

Tony2764, 2020-08-26:
Bonjour,

Oui j'ai effectué cette opération et ca n'a pas fonctionné.

Hello,

 Yes I did this operation and it didn't work.

Anthony

Anxiro, 2020-08-26:
1. Did you try the "solution" with a ELM adapter? Best to try is one **without** wifi, since your screen is black. It might be stuck in a bootloop.
2. Do you have a black screen but radio is working? Bluetooth working? If so, you tried to swipe the screen (in any direction)?

If you tried number one with USB ELM adapter and number two is not active, then I think you have to clone the firmware of a second radio (same type of car) and then flash it into yours. That's what I would do. But it's not an easy process if your device is bricked... I read people will usually get this problems using some kind of 'virgin mode'.

Tony2764, 2020-08-26:
My adapter is an obdlink sx. And seems well recognized. No the radio does
not work, nor the Bluetooth, only the ventilation works. As far as cloning
is concerned, it is beyond my skills ... Thanks a lot for your help.
Anthony.

shrlnm, 2020-08-26:
@Tony2764 I may try to help only after virgin MID/HIGH button or any other reset. I can not help after wrong upgrade.

Tony2764, 2020-08-29:
Good morning;
The problem is solved at this level. I was able to send a macro and everything is on! :-):-)
However, I no longer have a Gps map, there is a black wheel that toune.
It is not possible to install maps or update from v7 to V8.
Each time it is updated, it restarts, indicating that an error has occurred and that the current version is kept.
Before each reboot, there is a command line: RESETIPC193
[image]

Please help me out.

THANK YOU

Bonjour;
Le problème est résolu à ce niveau. J'ai pu envoyer une macro et tout c'est allumés! :-) :-)
Cependant, je n'ai plus de carte Gps, il y a une roue noire qui toune.
Il est impossible d'installer les cartes ou une mise à jour de v7 à V8.
À chaque mise à jour, il redémarre en précisant qu'une erreur est survenue et que la version actuelle est conservée.
Avant chaques redémarrage, il y a une ligne de commande : RESETIPC193
[image]

Aidez moi s'il vous plait .

MERCI

estyasser, 2020-08-29:
> Good morning;
> The problem is solved at this level. I was able to send a macro and everything is on! :-):-)
> However, I no longer have a Gps map, there is a black wheel that toune.
> It is not possible [...]

Hi Tony,
I had the same issue, upgraded from v2 to v7 and stuck on bootloop. I can access the r-link2 from the OBD and see the ecu's, but don't know how to fix the bootloop.
Since you were able to fix the bootloop, can you please tell me how you managed to do that? what's the macro you sent and how did you send it to the system?

I can also help in decompiling the firmware file and maybe we can figure a way to solve this for all.

Tony2764, 2020-08-29:
I was able to restore with the link below. With an elm327 not obdsx. That's why I didn't get there at first.
 
https://gitlab.com/py_ren/pyren/-/wikis/how-to-rlink2-black-screen

You have 4 macros available.
it worked for me. But my problem is not completely resolved. 
Do you have a solution to help me? 

good to you

estyasser, 2020-08-29:
I tried those macros before but didn't work for me either.
Was your issue a boot loop and got fixed with the maco files?

Unfortunately I'm still unable to get it working. I'm no expert or tech pro but I'm trying to find a solution.

If you have any tech background I can help with providing the decompiled software files, maybe we can reach something.

Anxiro, 2020-08-29:
Maybe I'm sounding stupid, just trying to help, thinking out loud. This sounds familiar to my friend who had a boot loop in Rlink-2. He unplugged his battery, made sure all power was gone (15-20min no battery). Then after plugging the battery back in, he hard reset the Rlink-2 by holding the power button for 30 seconds.

I think it's good to try the hard reset part first before trying unplugging the battery. Not sure if you need a radio code these days on the Rlink-2, but if you do, make sure you have it.

estyasser, 2020-08-29:
Thank you @Anxiro for the suggestion. Every thought counts and it's better to brainstorm the issue until we come up with a fix. 
I'll definitely give it a try and I'll update you.

Tony2764, 2020-08-29:
My problem is that I had a black screen, following reset settings.
I don't have a bootloop problem, just that my rlink no longer takes updates, it hangs during the update phase at the place where I posted the photo.
And restart with an installation failure message

shrlnm, 2020-08-29:
@Tony2764  what LGU version do you have now? Which one did you try to install?
@estyasser  I afraid your r-link 2 is dead ((. As I know the ciclic reboot con not be solved without re-flashing on the table

Tony2764, 2020-08-29:
I currently have version 7, i have tried installing v8 and v9 but it doesn't work.  It restarts at reset ipc as in the photo.  
What can I do to fix this problem?
Thank you

estyasser, 2020-08-29:
@Anxiro I tried unplugging the battery and hard resetting the screen but didn't work. 

@shrlnm so is there any I can reflash the firmware? Because the dealer here told me a new radio unit replacement will cost $3,100 😱

I have the firmware decompiled, meaning the recovery, system, kernal, boot , etc images are extracted. I don't know how this might help.

Karimou90, 2020-10-13:
Hello
 I have the same problem that you have I pressed virginMode or diagRw I don't know the screen is off but it turns on again I have the radio that's all I tried the technique with pyren it did not work I took everything apart and  everything unplugged nothing helps if someone has a solution I am taking
 Renault we try with their diag they also did not succeed

BillyJacobie, 2021-04-16:
I have been having this issue since 2019 when I updated my Rlink2 from 2.2.19.300 to v8
I must admit that I realize only now that it was my mistake because I did not do **enough** research and found out afterwards that my HARDWARE does not support V8, it never came on again, until today..

After 2 years of struggling I managed to restore my unit and brought it back to life from boot loop
I think it might have been lucky but I believe I found solution to units that have a green flash on the **Control Panel** that is stuck in the boot loop after upgrading from V2 to V7+.

I saw this post and it is the same as what I got, when i finally got it out the boot loop; 
https://github.com/cedricp/ddt4all/issues/430#issuecomment-683299637

@shrlnm I think there might be a solution to boot loop, but before I get anyone's hopes up maybe we can test solution?

spriti25, 2021-06-18:
@BillyJacobie can you please tell us how you got out of the Bootloop?

Klauzevitc, 2021-07-13:
@BillyJacobie can you please help me to fix bootloop ?

AlMeu, 2021-08-01:
I test me too and I can't revive a unit in boot loop.
Some other situations, yes.

AlMeu, 2021-08-01:
> Hello
> I have the same problem that you have I pressed virginMode or diagRw I don't know the screen is off but it turns on again I have the radio that's all I tried the technique with pyren it did [...]

I think I can help you, if you didn't solve it yet.

AlMeu, 2021-10-13:
> 
> I went to an official Renault garage. They updated my r-link 2 from V2 to V7 and it crashed.. Unfortunately they can't do anything for me, they offered me a new system for 1300euro, Wich is a crazy price.. At this moment my r-link is working but I have no sound at all anymore and the system seems slow.. Could you help me out sir? If so, you are a life saver!! grin

As you describe the situation is not good at all. Your unit was too old for this upgrade.
Unfortunately for you the only way is to change the unit.
That price for unit is crazy, I believe me too. With a unit clean, I might help you. If you want leave me an email and we speak.

karam-soliman, 2022-02-04:
@AlMeu I have faced the same but after update from 3 to 9, so  I would be interested too. akareem.soliman@gmail.com

gilnis2, 2022-12-26:
> I have been having this issue since 2019 when I updated my Rlink2 from 2.2.19.300 to v8 I must admit that I realize only now that it was my mistake because I did not do **enough** research and found out afterwards that my HARDWARE does not support V8, it never came on again, until today..
> 
> After 2 years of struggling I managed to restore my unit and brought it back to life from boot loop I think it might have been lucky but I believe I found solution to units that have a green flash on the **Control Panel** that is stuck in the boot loop after upgrading from V2 to V7+.
> 
> I saw this post and it is the same as what I got, when i finally got it out the boot loop; [#430 (comment)](https://github.com/cedricp/ddt4all/issues/430#issuecomment-683299637)
> 
> @shrlnm I think there might be a solution to boot loop, but before I get anyone's hopes up maybe we can test solution?

Hi @BillyJacobie , I'm in the same situation, how did you solve this boot loop?

trueAl, 2022-12-26:
> Hi @BillyJacobie , I'm in the same situation, how did you solve this boot loop?

Be careful, someone sent me a mail offering to sell a cheap new unit. It read a lot like scam, to say the least. 
In my case the dealer replaced the unit for free after I said my unit was defective after updating with usb stick from Website

AlMeu, 2022-12-26:
@trueAl 
Perhaps your case was because the car was still under warranty.
But as you see here speaking abou v2.2 units, that means cars between 2015, 2016 and even 2017 which warranty expired.
If you go to delaer with such car NO ONE will change the unit for free.
Moreover, if you go to dealer for upgrade from 2.2 to 7.0 with such car, they will put you to sign if upgrade fails it is under your responsability/charge. And guess which is the price for replace the unit at dealer .... with one new ...

gilnis2, 2022-12-26:
Thanks for the warning!
And I also went to the dealer and he said a new unit will cost some crazy price (something like 1000$).

It's just so hard to believe that because of a software (firmware) issue there's a need to replace the whole unit together with all the working components it has (Touch screen, hardware). It's really crazy there's not any option to just force flash the software on that unit.

AlMeu, 2022-12-26:
@gilnis2 
No need to change all navigation components. Need to change only the R-Link unit.
And of course at dealer prices are crazy.
I can help with new units at much better price or old units unlocked.

gilnis2, 2022-12-27:
@AlMeu 
But still the R-Link unit itself includes a lot of components that are still good. For example the touch screen itself, the CPU processor and much more.
It is very weird to replace hardware only because of defective software.

I would like maybe to buy an old unit, but unfortunately I'm from Israel and not from Europe so I think the delivery would be expensive, but thank you

AlMeu, 2022-12-27:
You have right, but when the CPU it's blocked because wrong upgrade, or memory it's out of order, there is nothing to do.
I respect your wish to revive it. Maybe you are more lucky then many others who tried.
I have tested mostly of all known methods and everything what cross my mind, but so far I don't find a solution.
First of all the flash for boot is protected for write ... so ...

ubem23, 2023-01-15:
> > Good morning;
> > The problem is solved at this level. I was able to send a macro and everything is on! :-):-)
> > However, I no longer have a Gps map, there is a black wheel that toune.
> > It is not possible to install maps or update from v7 to V8.
> > Each time it is updated, it restarts, indicating that an error has occurred and that the current version is kept.
> > Before each reboot, there is a command line: RESETIPC193
> > [image]
> > Please help me out.
> > THANK YOU
> > Bonjour;
> > Le problème est résolu à ce niveau. J'ai pu envoyer une macro et tout c'est allumés! :-) :-)
> > Cependant, je n'ai plus de carte Gps, il y a une roue noire qui toune.
> > Il est impossible d'installer les cartes ou une mise à jour de v7 à V8.
> > À chaque mise à jour, il redémarre en précisant qu'une erreur est survenue et que la version actuelle est conservée.
> > Avant chaques redémarrage, il y a une ligne de commande : RESETIPC193
> > [image]
> > Aidez moi s'il vous plait .
> > MERCI
> 
> Hi @Tony2764, I had the same issue, upgraded from v2 to v7 and stuck on bootloop. I can access the r-link2 from the OBD and see the ecu's, but don't know how to fix the bootloop. Since you were able to fix the bootloop, can you please tell me how you managed to do that? what's the macro you sent and how did you send it to the system?
> 
> I can also help in decompiling the firmware file and maybe we can figure a way to solve this for all.

hi guys i am in the same situation every fw i put it reboots 3 times with the same message how to have solved?

Tony2764, 2023-01-15:
Bonjour. 

Je n'ai jamais pu réparer ce problème. 
J'ai été contraint de remplacer le calculateur radio.
Bonne recherche

ubem23, 2023-01-15:
Je suppose que je suis forcé aussi, mais avez-vous acheté une radio d'occasion ou une nouvelle ?

AlMeu, 2023-01-15:
@ubem23 I already told you, in othr topic, that your unit have memory problem and fails in writting. 
If you buy old unit you get it blocked.
If you buy new you will not have problem.

ubem23, 2023-01-15:
@AlMeu but new you mean in the workshop Renault or also the ones on the internet? in the Renault they asked me 900€.

AlMeu, 2023-01-15:
If you want go to Renault workshop, you can go and pay those money what they asked you.
For sure they will solve your problem.

ubem23, 2023-01-15:
@AlMeu  you can help me I read in the posts above that you know how to do it

AlMeu, 2023-01-15:
I already answered you here: 
https://github.com/cedricp/ddt4all/issues/148#issuecomment-1383182172

ubem23, 2023-01-15:
@AlMeu ubem23@outlook.it

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

SG311288, 2024-11-30:
@shrlnm : Hello,
I have a similar issue. I performed an incorrect update and ended up with a black screen.
I have a Megane IV Sedan. I tried using PyREN but couldn't succeed—always getting "no data."
I would greatly appreciate it if you could contact me and help resolve this issue.

Thank you in advance,
arblitz88@gmail.com

Tony2764, 2024-11-30:
Bonjour.

Demande une configuration automatique du calculateurs radio chez Renault.

Cela prend 2 mn

Bonne journée
Anthony

SergeyG31, 2024-11-30:
Thank you, I was already there and they don't know how to handle the software, they offered me to change media. Of course I didn't agree until I find a solution :)
@Tony2764

Tony2764, 2024-11-30:
Il faut aller voir une autre concession.
Si ils changent le support, il devront faire obligatoirement une
configuration automatique

SergeyG31, 2024-11-30:
Is it possible to do it with pyren\dd4all?

abdelfr, 2025-01-02:
> As you describe the situation is not good at all. Your unit was too old for this upgrade.
>
> Unfortunately for you the only way is to change the unit.
>
> That price for unit is crazy, I believe me too. [...]

Hello sir

I have a crashed r link 2 and have a brand new virgin unit. Is there a chance you can help me to program it for my car ?