cedricp/ddt4all #432: Megane 3 - External Temperature Display on Cluster

johnroberts2k, 2020-08-08:
Does anyone know if it's possible to display the external temperature on the cluster on a Megane 3? If so, where do I find the config parameter? What about the clock and time too?

Marianpol, 2020-08-08:
[image]

charliehoward, 2020-08-14:
I would also like to know this, tried a few times to get it on my 265 but haven't got it working

AndyIreland, 2020-08-16:
I think after making a change you might have to press the #prise en ...~ button at the bottom 
[image]

charliehoward, 2020-08-16:
All of my other settings have worked fine, might try it again though

fparmentier, 2020-10-31:
This is not working on older cluster types.
Like mine, its from 2010 and doesnt' support it :(

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif