cedricp/ddt4all #433: ⫷Cornering dimming and speed⫸

JABICHOCU, 2020-08-09:
Hello to you all, this refers to Clio IV or Captur I  users. I´d like to investigate a bit on  the different options of changing Cornering parameters on these vehicles. I see that referring to dimming , I´ve seen cars configurations including  these two options>

★**Cxx_dimming_time_CF => "300" or  “220”**  What is the  difference  ???  Does anybody know??

And about speed at which Cornering stops working ,I´ve seen Clios and Capturs have the same>

★**Cxx_CorneringSpeed_CF => "40"**   Does anybody konw if  this can be changed? Have you tried??

Thanx everyone in advance who´d like to help, regards ㋛

JABICHOCU, 2020-08-13:
Anybody out there??? :) Pakoto, where are you???

gerzilov, 2021-05-04:
**Nbx_cornering_CF =>** cornering ON( 1 )/OFF( 0 )

**Cxx_A1_angle_CF =>** Angle of the steering wheel that turn ON cornering light

**Cxx_A2_angle_CF =>** Angle of the steering wheel that OFF cornering light. From A1 to A2 = dimming.

**Cxx_dimming_time_CF =>** time in 1/100 sec (300=3sec) after deactivation the turn signal but without turn the the cornering light goes to 0 ( 300 = 3sec dimming to 0 after turn signal off )  

**Cxx_MAX_power_PWM_ratio_CF =>** dimming from 0 to 1 eg 0.5 for 1/2 power

**Cxx_min_power_PWM_ratio_CF =>** dimming from 0 to 1 eg 0.2 for 1/5 power

**Cxx_CorneringSpeed_CF =>** the speed above cornering is off

Regards

ohayyam, 2021-05-04:
> **Nbx_cornering_CF =>** cornering ON( 1 )/OFF( 0 )
>
> **Cxx_A1_angle_CF =>** Angle of the steering wheel that turn ON cornering light
>
> **Cxx_A2_angle_CF =>** Angle of the steering wheel that OFF [...]

Where did you get these information ?

gerzilov, 2021-05-04:
Trial and error....Tested settings week ago on my captur 2019ph2 09 tce.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif