cedricp/ddt4all #434: ELM327 WiFi

johnroberts2k, 2020-08-11:
I've read about some Chinese clones of ELM327 not being compatible with ddt4all. Im not sure if my ELM327 WiFi is a clone or not. I've been using it for many years as a general scanner with iOS app called dashcommand.

My question is, how do I know if my device is compatible? If I connect my laptop to it and then if ddt4all shows the green connected icon, does that mean it's compatible? Or could it show connected but give random issues?

Thanks

KarelSvo, 2020-08-14:
An Elm test program is built into ddt4all. Why don't you use it, to verify your Elm?

rldchua, 2020-11-14:
Cash share what’s the test program?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif