cedricp/ddt4all #435: magnifying glass not finding ECU's with BT ELM327

AndyIreland, 2020-08-13:
I have just purchased a (cheap admittedly) Bluetooth ELM327 and when plug into my Clio mk4 OBD2 socket it lights up red then blue then green flashes a couple of times when I press the magnifying glass to search for ECU's . The bottom left green bar is 'connected' and after pressing the magnifying glass and selecting 'CAN' the green progress bar gets up to 100% but finds no ECU's in the window to the left. 

Here is a test I done with an ELM327 android app but I dont understand any of the results. Should it be compatible with DDT4all looking at its capabilities? 

Scan date=2020-08-13 17:18:50
Device name=OBDII
Device mac=86:DC:52:DD:F6:F1
Device Version (Declared)=ELM327 v2.1

Command	Version	Result
AT@1	1.0	OK
ATSPA6	1.0	OK
ATTPA6	1.0	OK
ATTP6	1.0	OK
ATSP00	1.3	OK
ATSP6	1.0	OK
ATAL	1.0	OK
ATNL	1.0	OK
ATAMC	2.0	KO
ATAMT20	2.0	KO
ATRA70	1.3	OK
ATSR70	1.2	OK
ATAR	1.2	OK
ATAT0	1.2	OK
ATAT2	1.2	OK
ATAT1	1.2	OK
ATBD	1.0	OK
ATBRT0F	1.2	OK
ATCAF0	1.0	OK
ATCAF1	1.0	OK
ATCEA04	1.4	OK
ATCERF1	2.2	KO
ATCEA	1.4	OK
ATCF00000111	1.0	OK
ATCF111	1.0	OK
ATCFC0	1.0	OK
ATCFC1	1.0	OK
ATCM00000111	1.0	OK
ATCM111	1.0	OK
ATCP18	1.0	OK
ATCRA7C0	1.3	OK
ATCRA000007C0	1.3	OK
ATCRA	1.4b	KO
ATCS	1.0	OK
ATCSM0	1.4b	KO
ATCSM1	1.4b	KO
ATCTM5	2.1	KO
ATCTM1	2.1	KO
ATCV0000	1.4	OK
ATD1	1.3	OK
ATD0	1.3	OK
ATDP	1.0	OK
ATDPN	1.0	OK
ATE0	1.0	OK
ATE1	1.0	OK
ATFCSD0430FF00	1.1	OK
ATFCSH000007B0	1.1	OK
ATFCSH7B0	1.1	OK
ATFCSM0	1.1	OK
ATFE	1.3a	OK
ATH0	1.0	OK
ATH1	1.0	OK
ATI	1.0	OK
ATIB10	1.0	OK
ATIB12	2.2	KO
ATIB15	2.2	KO
ATIB48	1.4	OK
ATIB96	1.0	OK
ATIFRS	1.2	OK
ATIFRH	1.2	OK
ATIFR0	1.2	OK
ATIFR2	1.2	OK
ATIFR1	1.2	OK
ATIFR6	2.2	KO
ATIFR5	2.2	KO
ATIFR4	2.2	KO
ATIGN	1.4	OK
ATIIA7A	1.2	OK
ATJS	1.3	KO
ATJE	1.3	KO
ATJHF0	1.4b	KO
ATJHF1	1.4b	KO
ATJTM5	1.4b	KO
ATJTM1	1.4b	KO
ATKW	1.3	OK
ATKW0	1.2	OK
ATKW1	1.2	OK
ATL0	1.0	OK
ATL1	1.0	OK
ATM1	1.0	OK
ATM0	1.0	OK
ATPPFFON	1.1	OK
ATPPFFOFF	1.1	OK
ATR0	1.0	OK
ATR1	1.0	OK
ATRV	1.0	OK
ATS0	1.3	OK
ATS1	1.3	OK
ATSDFF	1.4	OK
ATRD	1.4	OK
ATSH000007E0	2.0	KO
ATSHA848F1	1.0	OK
ATSH7B0	1.0	OK
ATST32	1.0	OK
ATSWFF	1.0	OK
ATTAF9	1.4	OK
ATV1	1.3	OK
ATV0	1.3	OK
ATWMC133F13E	1.0	OK
ATPC	1.0	OK
ATWS	1.0	OK
ATD	1.0	OK
ATZ	1.0	OK

[image]
[image]
[image]

[ELM327_Identifier_log1.txt](https://github.com/cedricp/ddt4all/files/5070400/ELM327_Identifier_log1.txt)

@cedricp  & others

KarelSvo, 2020-08-14:
An Elm test program is built into ddt4all. Why don't you use it, to verify your Elm?

AndyIreland, 2020-08-14:
> An Elm test program is built into ddt4all. Why don't you use it, to verify your Elm?

Thank you, good Idea I forgot about that . I will do

AndyIreland, 2020-08-15:
Here is the ELM benchmark I done through DDT4ALL:

v1.0 ATZ [OK]
v1.0 ATWS [OK]
v1.0 ATE1 [OK]
v1.0 ATL1 [OK]
v1.0 ATL0 [OK]
v1.0 ATI [OK]
v1.0 AT@1 [OK]
v1.0 ATRV [OK]
v1.3 ATS1 [OK]
v1.3 ATS0 [OK]
v1.0 ATCAF0 [OK]
v1.0 ATCFC0 [OK]
v1.0 ATH0 [OK]
v1.0 ATR1 [OK]
v1.2 ATKW0 [OK]
v1.3 ATD1 [OK]
v1.3 ATD0 [OK]
v1.0 ATAL [OK]
V1.0 ATSP 3 [OK]
V1.0 ATSP 4 [OK]
v1.2 ATIIA 01 [OK]
v1.4 ATSI [OK]
V1.0 ATSP 5 [OK]
v1.4 ATFI [OK]
V1.0 ATSP 6 [OK]
V1.0 ATSP 7 [OK]
V1.0 ATSP 8 [OK]
v1.0 ATPC [OK]
v1.0 ATSH 012 [OK]
v1.0 ATSW 96 [OK]
v1.0 ATSW 34 [OK]
v1.0 ATWM 817AF13E [OK]
v1.0 ATIB 10 [OK]
v1.0 ATST FF [OK]
v1.3 ATCRA 012 [OK]
v1.3 ATCRA 01234567 [OK]
v1.1 ATFC SH 012 [OK]
v1.1 ATFC SH 00112233 [OK]
v1.1 ATFC SD 300000 [OK]
v1.1 ATFC SM 1 [OK]
v1.2 ATAT0 [OK]
v1.2 ATAT1 [OK]
v1.0 ATCM 123 [OK]
v1.0 ATCM 12345678 [OK]
v1.0 ATCF 123 [OK]
v1.0 ATCF 12345678 [OK]
v1.4 ATMA [OK/TIMEOUT]
STN11xx TEST 1 STI [FAIL]
STN11xx TEST 2 STDI [FAIL]
STN11xx TEST 3 STPR [FAIL]
STN11xx TEST 4 STPBRR [FAIL]
STN11xx TEST 5 STPRS [FAIL]
Result: 47 succeeded from 52

For Media CAN (CAN Line 2) on STN1170, ELS27: Connect OBD pin 3 with 13 and 11 with 12.

AndyIreland, 2020-08-15:
And here is Log after I searched for ECU's with the magnifying glass:

>
 ATI                    ATI
ELM327 v2.1

>
 AT@1                   AT@1
OBDII to RS232 Interpreter

>
 ATRV                   ATRV
11.2V

>
 ATS1                   ATS1
OK

>
 ATS0                   ATS0
OK

>
 ATCAF0                 ATCAF0
OK

>
 ATCFC0                 ATCFC0
OK

>
 ATH0                   ATH0
OK

>
 ATR1                   ATR1
OK

>
 ATKW0                  ATKW0
OK

>
 ATD1                   ATD1
OK

>
 ATD0                   ATD0
OK

>
 ATAL                   ATAL
OK

>
 ATSP 3                 ATSP 3
OK

>
 ATSP 4                 ATSP 4
OK

>
 ATIIA 01               ATIIA 01
OK

>
 ATSI                   ATSI
BUS INIT: ...ERROR

>
 ATSP 5                 ATSP 5
OK

>
 ATFI                   ATFI
BUS INIT: ERROR

>
 ATSP 6                 ATSP 6
OK

>
 ATSP 7                 ATSP 7
OK

>
 ATSP 8                 ATSP 8
OK

>
 ATPC                   ATPC
OK

>
 ATSH 012               ATSH 012
OK

>
 ATSW 96                ATSW 96
OK

>
 ATSW 34                ATSW 34
OK

>
 ATWM 817AF13E          ATWM 817AF13E
OK

>
 ATIB 10                ATIB 10
OK

>
 ATST FF                ATST FF
OK

>
 ATCRA 012              ATCRA 012
OK

>
 ATCRA 01234567         ATCRA 01234567
OK

>
 ATFC SH 012            ATFC SH 012
OK

>
 ATFC SH 00112233       ATFC SH 00112233
OK

>
 ATFC SD 300000         ATFC SD 300000
OK

>
 ATFC SM 1              ATFC SM 1
OK

>
ATAT0                   ATAT0
OK

>
ATAT1                   ATAT1
OK

>
 ATCM 123               ATCM 123
OK

>
 ATCM 12345678          ATCM 12345678
OK

>
 ATCF 123               ATCF 123
OK

>
 ATCF 12345678          ATCF 12345678
OK

>
 ATMA                   ATMA
TIMEOUT
 STI               STI
?

>
 STDI              STDI
?

>
 STPR              STPR
?

>
 STPBRR            STPBRR
?

>
 STPRS             STPRS
?

>
Initilizing ELM with speed 115200...
Trying to open portCOM3 @ 115200
Checking port speed: 38400
Start COM speed : 38400
1580 loaded ECUs in database.
Warning, address 5D is not mapped
Scanning ECU INSULATOR
Scanning ECU ACC
Scanning ECU IDM
Scanning ECU BCM/UCH
Scanning ECU UPC-EMM
Scanning ECU 4WS
Scanning ECU CSHV
Scanning ECU HVAC
Warning, address 404 is not mapped
Warning, address 403 is not mapped
Warning, address 401 is not mapped
Warning, address 400 is not mapped
Warning, address 408 is not mapped
Warning, address 40A is not mapped
Scanning ECU Audio
Skipping CAN extended address (not supported yet)  2D
Scanning ECU UBP
Scanning ECU R RE
Scanning ECU ADP
Scanning ECU FRRAD
Scanning ECU Airbag
Scanning ECU Additional Heater
Scanning ECU MIU
Scanning ECU Navigation
Skipping CAN extended address (not supported yet)  16
Scanning ECU Cluster/TDB
Scanning ECU Tacho
Scanning ECU DLOCK
Scanning ECU 4WD
Scanning ECU TCU
Scanning ECU WCGS
Scanning ECU CAN Adapter
Scanning ECU VSP
Scanning ECU AAU
Scanning ECU C Box
Scanning ECU Suspension pilotee
Scanning ECU ABS/ESC
Scanning ECU EMCU
Scanning ECU HLS
Scanning ECU Power steering
Scanning ECU TPMS
Scanning ECU MC HEV FSCM
Scanning ECU EPS
Scanning ECU SVS
Scanning ECU SU
Scanning ECU SCRCM
Scanning ECU SCCM
Skipping CAN extended address (not supported yet)  E1
Skipping CAN extended address (not supported yet)  E0
Skipping CAN extended address (not supported yet)  E3
Skipping CAN extended address (not supported yet)  E2
Warning, address EE is not mapped
Scanning ECU Audio
Scanning ECU TCASE
Scanning ECU TPAD
Scanning ECU HU
Scanning ECU HFCK
Scanning ECU Frein de parking electrique
Scanning ECU Aide au parking
Scanning ECU S-DCDC
Skipping CAN extended address (not supported yet)  60
Scanning ECU AVM
Scanning ECU FCAM
Scanning ECU SL
Scanning ECU SOW Right
Scanning ECU VCCU
Scanning ECU BCB
Scanning ECU PEB
Scanning ECU HFM
Scanning ECU LBC (HEV)
Scanning ECU LBC (EV)
Scanning ECU (H) EVC
Scanning ECU PLC/PLGW
Scanning ECU ADAS-Sub
Scanning ECU BSW
Warning, address 6C is not mapped
Skipping CAN extended address (not supported yet)  15
Scanning ECU AT
Scanning ECU LDCM
Scanning ECU Magic Folding-RDCM
Skipping CAN extended address (not supported yet)  18
Skipping CAN extended address (not supported yet)  DF
Warning, address DD is not mapped
Scanning ECU BCB-OBC_29b
Skipping CAN extended address (not supported yet)  DB
Warning, address DC is not mapped
Skipping CAN extended address (not supported yet)  DA
Skipping CAN extended address (not supported yet)  7B
Scanning ECU ECM/Injection
Scanning ECU Engineering
Scanning ECU SR
Scanning ECU LKS
Skipping CAN extended address (not supported yet)  41
Scanning ECU DDCM
Scanning ECU PBD
Scanning ECU LBC2
Scanning ECU DCM Renault
Skipping CAN extended address (not supported yet)  73
Scanning ECU SCU
Scanning ECU HSG - LNG
Scanning ECU DCM
Warning, address 48 is not mapped
Skipping CAN extended address (not supported yet)  D6
Skipping CAN extended address (not supported yet)  D2
Scanning ECU UDM/UPCU

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836