cedricp/ddt4all #438: Display odo after ABS calculator Megane 3

aliloup85, 2020-08-15:
Hi all (sorry for my english)
I've a Megane 3, 1.9Dci (BZ0) with 230.000km. Ive changed the ABS calculator (situated behind the engine), but my odo is now 345000km. Is there any solution to display my real km on the dash (230.000km)?

Thank you all.

Salut à tous,
J'ai une mégane 3, 1.9dci de 2009 avec 230.000km sur le compteur. Après avoi changé le bloc ABS qui était hs, mon kilométrage est monté à 345000km. Y a-t-il une solution de reprogrammer mon bloc abs afin d'afficher mon kilométrage réel.

Merci à vous tous.

Marianpol, 2020-08-18:
Odometer and a ABS calculator are connected in term of a mileage. If some of these two modules connected to a CAN network of a car has higher mileage, the another one saves it in the memory. 
So, now your odometer has ABS module mileage. 
You have to correct a mileage in an EEPROM memory of both modules independently.
Be sure to not simultaneously connect both modules to the car if some of these has a higher mileage.

aliloup85, 2020-08-18:
> Odometer and a ABS calculator are connected in term of a mileage. If some of these two modules connected to a CAN network of a car has higher mileage, the another one saves it in the memory.
> So, now [...]

What the mechanic has done, is to test the other abs (with 345k km) just by placing the wires, a'd after he has connected mine as it was before. When I took it to scan they said to me that my car has actually the informations of another car (vin and mileage in memory). They said to me that they have to:
1. ask for a file from the manufacturer Renault
2. configure the new abs on 0km
3. After placing the new abs module, they will have to "clean" the memory car of all those information and set them all over again to the original ones.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif