cedricp/ddt4all #439: No Data - Clio 4

MouadSoufiani, 2020-08-15:
Hello, 

I got no issues with the founded ECU (only 4). But when I check manually in the list I ogt ecu's with no data on it. The ELM is working because I can cahnge some parameteres in the red en green founded ecu's (clock display & temperature).

What could be the problem?

[image]

MouadSoufiani, 2020-08-15:
I was searching again and I found this. Could this be the problem? Can I just set it on 'with' and cahnge the parameters then?
[image]

AndyIreland, 2020-08-16:
4 ecus's does not seem a lot, I would have thought you had a lot more. Do you have auto wipers and auto headlights on your clio 4 ? - that EMM setting is just in the dashboard so changing that i shouldnt have thought would help ddt4all in finding more ECUs ... if the ECU's are not fitted to your car they are not fitted! in DDT4all other screens there are settings such as 'Lighting by EMM' and "wipers by EMM' am sure I have seen those parameters somewhere

AndyIreland, 2020-08-16:
there is a 'lighting by emm' parameter in the body control module (BCM) you could see if that is set to true or false I suppose - but be careful because if lights are not controlled by emm you might break them altogether by setting this setting 

[image]

MouadSoufiani, 2020-08-16:
Hello Andy,

Nice to see you replying.  I am following you on your blog also. You are doing a great job!

The setting that you said 'lighting by EMM' is on false. I don't have the auto lighting and auto wipers.

AndyIreland, 2020-08-20:
> Hello Andy,
> 
> Nice to see you replying. I am following you on your blog also. You are doing a great job!
> 
> The setting that you said 'lighting by EMM' is on false. I don't have the auto lighting and auto wipers.

Thanks for the compliment , glad you like the blog . That must be it then that the EMM is on false then because you dont have auto lighting and auto wipers they must be controlled in other ways without any ECU's

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif