cedricp/ddt4all #440: Help please!! How to release the rear pads (Electric handbrake) of Megane 3?

aliloup85, 2020-08-17:
please, someone can help me to activate the maintenance mode on rear brake pads on Megane 3 with DDT4ALL? 

Thanks a lot.

BarektE, 2020-08-18:
gönder mail adresini tarif edeyim

aliloup85, 2020-08-18:
> gönder mail adresini tarif edeyim

Mouhammedali on my Gmail address. Thanks a lot.

BarektE, 2020-08-18:
mouhammedali@gmail.com adresine nasıl yapacağınızın tarifini gönderdim

BarektE, 2020-08-18:
Where I marked the number 1, press start, the lining will be released, after replacing the lining, the number 2 is pressing the start.

aliloup85, 2020-08-18:
> Where I marked the number 1, press start, the lining will be released, after replacing the lining, the number 2 is pressing the start.

Thanks so much 🙏🙏🙏🙏🙏

avramg, 2020-08-28:
Hi can send to me also the procedure
thanx 
avram_gigel@yahoo.com

Best Regards

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif