cedricp/ddt4all #443: Search/reading does not return complete set of options

DrFsn, 2020-08-25:
Hi all,

I would like to ask for some help, trick, what can be the problem or what am I doing wrong ... The base problem is, that once I click on magnifying glass, in the middle section, I do not receive all the options as I should it. Two different ELMs were used, same result. 

I managed to activate Android auto without any problems, but I have to manually chose car, ‘Navigation’ - ‘MFD v4.6’ - ‘[ Navigation ] MFD v4.6’ - ‘Configuration’ ‘13 - ECU Configuration’. For other setting, like EMM, after manual selection it returns no data. Expected, as also magnifying glass does not find it in previous step …

Logs and PrtScrs attached. Thanks for any help. 
Dan

[image]
[image]
[log_file.txt](https://github.com/cedricp/ddt4all/files/5123893/log_file.txt)

DrFsn, 2020-08-25:
And this is how the search result should look like (marked with green on some PrtScr from web) …

[image]

BarektE, 2020-08-26:
You are probably opening the ddt4all program in a demo jacket.

[image]

DrFsn, 2020-08-26:
Thx, but nope, this is not the case. I used Connected mode and as I explain above, I manage to change Android Auto settings … Any other idea, are there any more useful logs to check?

xmenmaxx, 2020-08-27:
Hello,

I also have the same problem with a Kadjar 2016, I tried with two different OBD v1.5 interfaces and the result is the same, I don't realize what it could be, I use the latest database that is installed with ddt4all 5.5.0.
#cedricp, can you help us please?
Thanks in advance!
[image]
[image]

DrFsn, 2020-08-27:
Can be engine related? What kind of engine do you have @xmenmaxx? Mine is 1.6 dci, 130 PS ...

xmenmaxx, 2020-08-27:
> Can be engine related? What kind of engine do you have @xmenmaxx? Mine is 1.6 dci, 130 PS ...

I also have 1.6 dci, 4x4, 130 hp

KarelSvo, 2020-08-27:
> Hello,
> 
> I also have the same problem with a Kadjar 2016, I tried with two different OBD v1.5 interfaces and the result is the same, I don't realize what it could be, I use the latest database that is installed with ddt4all 5.5.0.
 

The latest Ecu database is from July 2020. But Renault does not update the Autoident data in the Ecu files.
Therefore there is no match for many ecus.

xmenmaxx, 2020-08-27:
> For green results you must to add the data manually.

How can I do this? I am a beginner in using ddt4all ....

xmenmaxx, 2020-08-27:
> The latest Ecu database is from July 2020.

@KarelSvo  where can I find the latest database?

DrFsn, 2020-08-27:
The point (in my case) is, that there are no or at least not all of expected “green results”. After manual navigating to some section, there is "no data" information. As I understand expected, as even “search” did not find it …

Not sure about database, my car is model year 2015, this engine is available at least from 2013 and my database is 2019, so that need to be covered or is there any other explanation?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif