cedricp/ddt4all #448: ddt4all clio 5

turning3533, 2020-08-29:
how can you use ddt4all on renault clio 5 despite having inserted the 'central gateway' control unit in this model? this control unit allows the reading of all the parameters but prevents the writing with ddt4all how can you pass it?

Marianpol, 2020-08-29:
You can't pass it right now. Maybe in the future someone will develop a gateway bypass like one used in FCA cars.
I think, here you can buy token that allows you to unlock a module (see EURO 5/EURO 6 Tab).
https://asos.renault.com/#/presentation

turning3533, 2020-08-29:
thanks for the answer ... the tocken you are talking about works only on renault clip interface right? is there no way to use it with ddt4all to make changes?

turning3533, 2020-08-29:
> 
> 
> You can't pass it right now. Maybe in the future someone will develop a gateway bypass like one used in FCA cars.
> I think, here you can buy token that allows you to unlock a module (see EURO 5/EURO 6 Tab).
> https://asos.renault.com/#/presentation

closed 2020-08-29 by turning3533