cedricp/ddt4all #450: Installing a ECU from one car to another

johnroberts2k, 2020-09-01:
When installing a used ECU like a UCH or UPC or DCM etc, when querying that ECU with CLIP it actually shows the original VIN number of the car that it came from. Although it works and everything the VIN seems to be programmed into every ECU. Just wondering if there is a way in ddt4all to change a vin on these ECU's so I can have it matching my car?

shrlnm, 2020-09-02:
@johnroberts2k not all ECU allows to change their VIN. Some of them require special configuration reset before changing

johnroberts2k, 2020-09-02:
@shrlnm I am looking at the UPC ECU, I swapped out my UPC for another one from a salvage scrap. The new one seems to be working and everything but if I read the VIN number from the configuration of that ECU it displays the other cars VIN number. Do you know if the UPC VIN number can be changed?

cedricp, 2020-09-02:
Which car and ecu model is it?

johnroberts2k, 2020-09-02:
> Which car and ecu model is it?

@cedricp it's a Megane 3, the ECU is "UPC - USM 95 BDKE - Sot C2/C3/C4 v3" that's the one that gets detected as green when I scan or ECU's

cedricp, 2020-09-03:
If you use DDT4All in edit mode, you can find the write VIN command : 
[image]
You can see '3B81' is the command to do it, replace the remaining zeros with you VIN/CRC (I think you can get it from another ECU, or recalculate it)
Then, with the manual command button (0x), write your car's VIN.
[image]

pakoto, 2020-09-03:
Renolink have a separate option to change VIN to ECUS. Is difficult to add into ddt4all? Thanks

cedricp, 2020-09-03:
That could be possible, but I'm very busy with other stuff right now to improve it.

pakoto, 2020-09-03:
[image]

johnroberts2k, 2020-09-03:
@cedricp thanks for that. I am fairly new to ddt4all. I've been able to make configuration changes but the instructions above i am a bit lost. You mentioned to run ddt4all in edit mode, is that the expert mode? I don't see any of those tabs like Request etc, not sure if its because i'm running in Demo mode outside the live environment. And what exactly is the CRC?

cedricp, 2020-09-04:
You can start the app with 2 modes (connected or edition), maybe you're running an old version...
The CRC is a code generated from VIN to ensure its validity. Give me your VIN, I'll send you the command to send.

johnroberts2k, 2020-09-05:
@cedricp the only options I see when DDT4ALL is loaded is Connected Mode, Demo and ELM Benchmark. I don't see edition. My version says: Version 5.4.0 ALL CARS Edition (Database 11/02/2019). Is this out of date version and what about the database version? Can I email me my VIN as I don't feel comfortable putting it on a public chat? Thanks.

BarektE, 2020-09-10:
[image]

TA1988, 2023-01-10:
Hi, I need to replace a faulty UPC ECU with a used one on a Renault Captur (X87): which ECU should I use to rewrite the VIN and CRC? Thank you!

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

aslanbey8371, 2025-06-19:
> [image]

hi help crc code