cedricp/ddt4all #451: Cannot find HANDFREE_DISABLE_AFTER_RKE_CF setting for Renault Zoe

AndyIreland, 2020-09-06:
Hello , I dont know if I am looking in the wrong place or just cannot find it or its just not there but I cannot find the HANDFREE_DISABLE_AFTER_RKE_CF setting for my 2013 Renault Zoe - have looked all in doorlock area and BCM and cannot find it. 
I want to be able to open the doors always by handsfree and not have the feature break when I use button on keycard . Managed to do it on a Clio mk4 but cannot find it for the renault zoe. 
Any ideas please?

BarektE, 2020-09-10:
Я сподіваюся, це працює для вас. Якщо ви не можете знайти блоки, я надішлю їх електронною поштою

[image]

AndyIreland, 2020-09-16:
> Я сподіваюся, це працює для вас. Якщо ви не можете знайти блоки, я надішлю їх електронною поштою
> 
> це не спрацює, якщо немає антени.
> 
> [image]

Thank you @BarektE for the reply - yes I found that setting when I was trying and it was set to 'With' already ... but it still breaks the handsfree feature if I choose to open or lock the door with the keycard button until I put the keycard in the slot and start the engine to reset handsfree again. 
In clio mk4 ecu it was a setting saying something like "disable RKE after button press" and then i chose 'disabled' but I just cannot find it in any of the Zoe ecu's

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

xspawn86, 2023-11-01:
If you use Ecu tweaker try  "Hand Free Module / Access By Switch / Access By Switch - Configuration"