cedricp/ddt4all #457: Who writes the DDT2000 database I need to contact them?

AndyIreland, 2020-09-17:
Just wondering who writes the actual DDT2000 Database so that the parameters can be changed on the ECU's ? 

I have 2 Renault cars at the moment a Clio Mk4 and a Renault Zoe electric vehicle. 

On the cluster for the Clio Mk4   Cluster X98_X87_X82_xHA_X98ph2_X87ph2_xHAph2_v8.02 Cluster/TDB
I can change various parameters such as disable rear seatbelt reminder on dashboard cluster and Disable EPS warning on Dashboard cluster among other things .

On the Renault Zoe cluster setting    TdB_X10_V1.12 Cluster/TDB  I want to disable those 2 warnings , but apart from the fact that the cluster parameter settings in the x10 Zoe is all written in french, in DDT4ALL I cannot find any settings to disable the EPS and rear seatbelt warnings in the parameters for the dashboard cluster of the Zoe .  - the Zoe cluster has a totally TFT screen by the way and the Clio mk4 has dials and a matrix screen. 

I was going to try and take a chance to disable EPS and rear seatbelt warning using the parameters of the  X98_X87_X82_xHA_X98ph2_X87ph2_xHAph2_v8.02 Cluster/TDB for the Zoe but decided against because it could damage the ecu or cluster for the zoe using the wrong ECU settings I should imagine. 

any ideas ? thanks. 
@cedricp and others 

Here is Cluster TDB settings for Renault Zoe EV:
[image]

Here is setting for Clio Mk4 
[image]

[image]

KarelSvo, 2020-09-17:
Config Generale_Inhib SBR_av
ON -  Buzzer SBR avant non inhibé (par défaut)
OFF - Buzzer SBR avant inhibe

AndyIreland, 2020-09-18:
> Config Generale_Inhib SBR_av
> ON - Buzzer SBR avant non inhibé (par défaut)
> OFF - Buzzer SBR avant inhibe

Thank you - but that just turns off the buzzer - it does not get rid of the message on the cluster saying '0 rear sealtbelts connected'

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

manio, 2024-02-10:
@AndyIreland 
> > Config Generale_Inhib SBR_av
> > ON - Buzzer SBR avant non inhibé (par défaut)
> > OFF - Buzzer SBR avant inhibe
> 
> Thank you - but that just turns off the buzzer - it does not get rid of the message on the cluster saying '0 rear sealtbelts connected'

Andy, I want to do the same change on ZOE (inhibit the buzzer) also I also has this in french.
Some questions - maybe you remember and can tell me:
1. How to prepare? Do i need to start the engine (brake pedal + P), or this is irrelevant ? Is there some specific procedure before writing the "Meter cluster" settings?
2. I assume that I need to click yellow button 'Inhibit SBR av' after a value change, so this single setting is saved, but what next - is the "Prise en compte des configurations" sufficient, or I also need to click "Retour a la config  par default" at the very bottom ?

AndyIreland, 2024-02-10:
@manio  - here you go:

just do it with car not ready ('engine' off) and keycard in slot
when you press Prise en compte des configurations  dash will briefly go dark/blank and you may get a few error messages briefly like ' Check ESC/HILL START/ABS' - dont worry about that , as long as the dash lights up again you should be OK. 
** DO AT YOUR OWN RISK**

[image]

[image]

manio, 2024-02-11:
Thank you very much, Andy! :)