cedricp/ddt4all #458: Add new vehicle models to DDT4All

pakoto, 2020-09-20:
Need to add new models to ddt4all.py with new DDT2000 ecus version

<Comment>All ECUs</Comment>
<Date>2020-07-10T14:32:18</Date>
<Projects><Almera/><AS1/><Edison/><H45/><HFE/><HFEPh2/><KJA/><L21B/><LZ2A/><March/><Nova/><p32/><p33A/><p33B/><P42Q/><P42R/><Primera/><PY1B/><PZ1A/><RF90/><u60/><VS11/><w176/><w205/><x02E/><x06/><x07/><x07Ph2/><x09/><x10/><x10Ph2/><x13A/><x21B/><x24/><x33/><x35/><x38/><x43/><x44/><x44Ph2/><x47/><x52/><x54/><x55/><x56/><x60A/><x60B/><x61/><x61Ph2/><x62/><x62Ph2/><x64/><x65/><x66/><x67/><x70/><x70Ph3/><x73/><x73Ph2/><x74/><x74Ph2/><x76/><x77/><x77Ph2/><x79/><x79Ph2/><x81/><x81Ph2/><x81Ph3/><x81Ph4/><x82/><x82Ph2/><x83/><x83Ph2/><x83Ph3/><x84/><x84ABSonly/><x84bugABS/><x84Ph2/><x85/><x87/><x87Ph2/><x89/><x90/><x91/><x91Ph2/><x91Ph3/><x92/><x94/><x95/><x95Ph2/><x96/><x98/><x98Ph2/><xBA/><xBB/><xBC/><xBG/><xBJ/><xCB/><xCC/><xCD/><xDC/><xEF/><xFA/><xFAPh2/><xFB/><xFBPh2/><xFC/><xFCPh2/><xFD/><xFDPh2/><xFF/><xFFPh2/><xFG/><xFI/><xFJ/><xFK/><xGA/><xGF/><xHA/><xHAPh2/><xHC/><xHN/><xJA/><xJB/><xJC/><xJD/><xJE/><xJF/><xJH/><xJI/><xJK/><xJL/><xJN/><xJO/><xMW/><xMZ/><xND/><xNE/><xNF/><xNN/><xPIV_C1A/><xPIV_C1AHS/><xZG/><xZGPh2/><xZH/><xZHPh2/><xZI/><xZJ/><xZJPh2/></Projects>

pakoto, 2020-09-20:
`

Twingo | x06
-- | --
Twingo III | x07
Twingo III Ph2 | x07Ph2
Twizy | x09
Zoe | x10
[Zoe] | xCB
Zoe (C1A-Neo) | x10Ph2
Wind | x33
Twingo II | x44
Twingo II Phase2 | x44Ph2
Kwid | xBA
Kwid EV | xBG
Kwid BR | xBB
Symbol / Thalia | x35
Clio II | x65
New Captur India | xBC
Modus | x77
Modus Phase2 | x77Ph2
Captur | x87
Captur Phase 2 | x87Ph2
Clio III | x85
Clio IV | x98
Clio IV Phase 2 | x98Ph2
Clio (C1A) | xJA
C-Hatch China (C1A) | xFI
Megane & Scenic | x64
Logan III Badge Renault | xJF
New EV (C1A HS Evo Sweet400) | xCC
New EV China version (C1A HS Evo Sweet400) | xCD
Megane II | x84
Megane II hors ABS | x84bugABS
Megane II only ABS | x84ABSonly
Megane II Phase 2 | x84Ph2
Megane III | x95
Megane III Phase 2 | x95Ph2
Fluence | x38
Fluence - Korea | xJL
xJH | xJH
Scenic IV | xFA
Scenic IV phase2 | xFAPh2
Megane IV | xFB
Megane IV Ph2 | xFBPh2
Kadjar Ph2 | HFEPh2
[SUV] | xHN
[SUV]Chine | xHC
Kadjar | HFE
Megane IV - Sedan | xFF
Megane IV - Sedan Ph2 | xFFPh2
(x38_Chine) | xFJ
Alpine A110 | xEF
New Captur (C1A) | xJB
New Captur Chine) | xJE
Kadjar CN | xZH
Kadjar CN Ph2 | xZHPh2
Laguna | x56
Avantime | x66
Koleos II | xZG
Kadjar Rus | xZI
Koleos II Ph2 | xZGPh2
Koleos II - Chine | xZJ
Koleos II - Chine Ph2 | xZJPh2
Laguna II | x74
Laguna II Phase 2 | x74Ph2
Laguna III Tricorps | x47
Laguna III | x91
Laguna III Phase 2 | x91Ph2
Laguna III Phase 3 | x91Ph3
(X96) | x96
Talisman | xFD
Talisman Phase II | xFDPh2
Latitude | x43
Safrane | x54
VelSatis | x73
VelSatis Phase 2 | x73Ph2
Espace IV | x81
Espace IV Phase 2 | x81Ph2
Espace IV Phase 3 | x81Ph3
Espace IV Phase 4 | x81Ph4
(x94) | x94
Espace V | xFC
Espace V Ph2 | xFCPh2
(xFG) | xFG
Master II | x70
Master II Phase 3 | x70Ph3
Kangoo | x76
New Kangoo | xFK
Docker II | xJK
Kangoo II | x61
Alaskan - (u55, xND) | u60
Kangoo II Phase 2 | x61Ph2
Trafic III | x82
Trafic III Phase2 | x82Ph2
Trafic II | x83
Trafic II Phase 2 | x83Ph2
Trafic II Phase 3 | x83Ph3
Master III | x62
Master III Phase 2 | x62Ph2
Master Chine | xDC
Mascott | x24
(X89) | x89

`

pakoto, 2020-09-20:
`

Logan | x52
-- | --
Logan III | xJI
Lodgy | x92
Kaptur - Captur (BAR/IN/RU) | xHA
Kaptur - Captur (BAR/IN/RU) Ph2 | xHAPh2
BM Lada | xGA
Duster | x79
Duster Phase 2 | x79Ph2
Arkana | xJC
Duster Phase 3 | xJD
Logan / Sandero | x90
Nova | Nova
Docker - Kangoo | x67

`

pakoto, 2020-09-20:
I see you @cedricp  add vehicles into ddt4all.py manually, and renolink reads /vehicles/projects.xml from ddt2000 folder. Is correct? Could be implemented?
[projects.zip](https://github.com/cedricp/ddt4all/files/5252126/projects.zip)

okankalfa, 2021-12-26:
so how to add ?

KarelSvo, 2021-12-27:
Simply add it yourself in ddt4all.py  https://github.com/cedricp/ddt4all/blob/master/ddt4all.py#L88

AndyIreland, 2021-12-27:
I am just curious how my members user name keeps popping up throughout the DDT4ALL.PY file when I open it up in txt editor? 

[image]

KarelSvo, 2021-12-27:
What you are showing, is not the content of the original ddt4all.py file.

AndyIreland, 2021-12-27:
> What you are showing, is not the content of the original ddt4all.py file.

I downloaded the DDT4all.py file from here github then opened with notepad

KarelSvo, 2022-01-01:
There is nothing in the lines from your link of what you show in your photo.  I have no idea what you have opened, definitely not the text content of the ddt4all Python file.
In your picture you can see an html coding not Python.

AndyIreland, 2022-01-01:
> There is nothing in the lines from your link of what you show in your photo. I have no idea what you have opened, definitely not the text content of the ddt4all Python file. In your picture you can see an html coding not Python.

you try downloading ddt4all.py from https://github.com/cedricp/ddt4all and then open it up in windows notepad - thats what i did and then thats what i got

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif