cedricp/ddt4all #459: Modification for new EasyLink

igno77, 2020-09-22:
If I have understood, in this moment, it's not possible to save modification on the new EasyLink (Captur II, Clio V and Megane ph2) due to a limitation of a gateway ecu. Someone have tried it? Did you find a way for write on it?
Thanks

jjaemny, 2020-10-02:
I have the same problem. Have you solved the problem?

KarelSvo, 2020-10-04:
You need to remove the glove box in order to establish a direct connection to CANH and CANL of the Easylink Ecu.
CANH pin 18, 19 and 20
CANL pin 38, 39 and 40
[image]

igno77, 2020-10-04:
What's you mean for "glove box"? Never seen before a can with 3 signal for each channel....

KarelSvo, 2020-10-04:
The Easylink Ecu is located above, behind the glove compartment of the Clio and Captur.

jjaemny, 2020-10-04:
If I have to use elm327 to modify the ecu equipped with an easylink system, should I connect elm327 to 38-40 pin?

cetinyusuf, 2021-03-15:
clio 5 CANH VE CANL nerede tam olarak obd soketine nasıl bağlantı yapabilirm güvenlik ğeçidini geçebirmiyim @KarelSvo

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif