cedricp/ddt4all #460: Enable remote lighting - Renault Captur 2016

Deekel, 2020-09-24:
I'm wondering if I can enable remote lighting to use with my light button on my keycard. My Captur is running Edison V13.... I know there has been issues enabling certain features for my particular BCM version. I'm just wondering if anyone has successfully tried to enable remote lighting with versions below 15?

EXcomZkko, 2020-09-25:
There is no option named enable remote lighting on V13 if I remember correctly. So, looks like it's impossible to activate I think.

cdemi, 2020-10-03:
Oh man I have looked so hard for this for my Captur. Do you think it's not possible or is it just missing in the configuration?

EXcomZkko, 2020-10-04:
> Oh man I have looked so hard for this for my Captur. Do you think it's not possible or is it just missing in the configuration?

Actually the answer is hidden in the question. It's not possible, because of that ECU does not supports this feature and does not have config option because of this.

You can try to look via v15-16 ECU file (can be selected manually) which has that option. But you will see "NODATA" word for the reason what I wrote.

So, sorry mate. :|

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif