cedricp/ddt4all #461: ddt4all "Program error" with "Stop" light on, Renault Talisman

dadoblu2000, 2020-09-25:
Hello everyone I write here hoping that you can help me solve a problem that I combined with ddt4all, in practice I touched some value that I should not touch, I wanted to reset the errors that were displayed but now the dashboard displays progra error and the warning light stop on, in addition, the language set is French and I cannot change it, while the other settings can be changed, the car works regularly, which is why I went to a renault workshop in my area and it informed me that a new one must be reprogrammed. ECU, I was not told which one it is and that I have to replace the front lambda probe, total intervention value 700 euros, I hope you can help me solve it on your own, the model is Talisman 1.6 130CV, here are some photos
[image]
[image]
[image]
[image]

pakoto, 2020-09-25:
err prog means that you enable an option into cluster that not enabled into ecus, or viceversa.

i.e. disable start-stop or tpms into cluster, when is enabled into BCM.

dadoblu2000, 2020-09-25:
is there a method to be able to tipristinsre how it was before?  between us I do not know what I could have activated / deactivated, but I have noticed that when I activate when I activate the high beams in automatic mode they no longer work, in addition to the settings that appear in the instrument panel when I try to change the language from French to Italian the  command takes it for an idtante but then returns to the French language ... can you tell me which control unit to check?

dadoblu2000, 2020-09-28:
thanks for your information, I solved it by configuring the Conf_MPU scheme by setting everything to absent all, now the spy is gone.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif