cedricp/ddt4all #462: Cedricp DDT4ALL

BarektE, 2020-09-28:
Mr. Cedricp Do you have any new work on DDT4ALL? if there is, how will we pack it? Thanks Zmiterm was packing new software and we were using it without any mistakes. I think we lost Zmiterm in Covid 19, it does not appear anywhere. We need someone more like him to make our job easier.

[image]

pakoto, 2020-09-28:
I packed last version using last 07.2020 database, modifying car list and bypassing expert mode and security warning: no need to check "old man icon" before write changes. 

This unofficial version is informed to Cedric, but not supported here in github.

https://dacieros.forumfree.it/?t=77922033

BarektE, 2020-09-28:
> Son sürümü 07.2020 veritabanını kullanarak, araç listesini değiştirerek ve uzman modunu ve güvenlik uyarısını atlayarak paketledim: değişiklikleri yazmadan önce "yaşlı adam simgesini" kontrol etmeye gerek yok.
> 
> Bu resmi olmayan sürüm Cedric'e bildirilir, ancak burada github'da desteklenmemektedir.
> 
> https://dacieros.forumfree.it/?t=77922033

The addition of different languages ​​is great. But languages ​​are missing. Is one of the largest markets throughout the world as the Renault car project has made Turkey the addition of Turkish missing. For me, the old man's getting up was not good for security.

pakoto, 2020-09-28:
I could make new version "with old man". Turkey installation messages could be easily added. If not exist in-app translation, you could modify spanish file replacing messages and we could add translation.

[ddt4all.zip](https://github.com/cedricp/ddt4all/files/5292238/ddt4all.zip)

BarektE, 2020-09-28:
> "Yaşlı adamla" yeni versiyon yapabilirim. Türkiye kurulum mesajları kolaylıkla eklenebilir. Uygulama içi çeviri yoksa, İspanyolca dosyasını mesajları değiştirerek değiştirebilirsiniz ve biz çeviri ekleyebiliriz.
> 
> [ddt4all.zip](https://github.com/cedricp/ddt4all/files/5292238/ddt4all.zip)

now i'm installing ddt4all you edited, let me check

pakoto, 2020-09-28:
.po file i sent now can be opened with notepad, and replace all spanish messages with your language. Don´t touch english messages.

BarektE, 2020-09-28:
ok

pakoto, 2020-09-28:
you can take other language .po file from /locale if is easier for you to translate it.

BarektE, 2020-09-28:
I am translating, I will send you when I translate

BarektE, 2020-09-28:
Lo traduje al turco. 

¿Cómo agrego este archivo a ddt4all? 

28/09/2020 15:41, pakoto yazmış:

> I could make new version "with old man". Turkey installation messages could be easily added. If not exist in-app translation, you could modify spanish file replacing messages and we could add translation. 
> 
> ddt4all.zip [1] 

 

Links:
------
[1] https://github.com/cedricp/ddt4all/files/5292238/ddt4all.zip
[2] https://github.com/cedricp/ddt4all/issues/462#issuecomment-699981273
[3]
https://github.com/

BarektE, 2020-09-28:
I translated the file I sent you

pakoto, 2020-09-28:
only drag file into a comment.

BarektE, 2020-09-28:
does not install the zip file

pakoto, 2020-09-28:
i can´t understand. Only need to zip your .po modified file an drag into comment

File .po don´t work with ddt4all, need to be convert. I can do it.

BarektE, 2020-09-28:
[ddt4all Turkish.zip](https://github.com/cedricp/ddt4all/files/5293191/ddt4all.Turkish.zip)

pakoto, 2020-09-28:
I see file without translate

msgid "Goto data"
msgstr "Ir a Datos"

msgid "Remove"
msgstr "Eliminar"

msgid "Add to screen"
msgstr "Añadir pantalla"
.
.
.

i suppose that would be something like this

msgid "Goto data"
msgstr "verilere git"

msgid "Remove"
msgstr "Kaldırmak"

msgid "Add to screen"
msgstr "bir ekran ekle"

(i´m using google translate)

BarektE, 2020-09-28:
[image]

pakoto, 2020-09-28:
ok, you only need to translate plugin messages? You could translate whole file and Cedric adds as language.

BarektE, 2020-09-28:
Yes

BarektE, 2020-09-28:
> tamam, sadece eklenti mesajlarını çevirmeniz mi gerekiyor? Tüm dosyayı çevirebilirsiniz ve Cedric dil olarak ekler.

I will translate them all

BarektE, 2020-09-28:
> tamam, sadece eklenti mesajlarını çevirmeniz mi gerekiyor? Tüm dosyayı çevirebilirsiniz ve Cedric dil olarak ekler.

Tamamlandığında dosyayı e-posta ile göndereceğim

pakoto, 2020-09-28:
You unzip into /locale and test. I don´t know if Cedric needs to modify ddt4all to add tr language.

[tr.zip](https://github.com/cedricp/ddt4all/files/5293738/tr.zip)

BarektE, 2020-09-28:
> / Locale içine açıp test edersiniz. Cedric'in tr dilini eklemek için ddt4all'ı değiştirmesi gerekip gerekmediğini bilmiyorum.
> 
> [tr.zip](https://github.com/cedricp/ddt4all/files/5293738/tr.zip)

Cedric had told me before and had sent me the file of the ecu tweaker translation so that I would add it if you translate it, and I had it translated to someone else. Now I'm turning it myself

BarektE, 2020-09-28:
I turned most of them already, there is little left

pakoto, 2020-09-28:
i sent only for test, if reads your first translated messages, I don´t know if ddt4all detects language code from windows, and if it´s TR or TK.

BarektE, 2020-09-28:
Türkçe tr

BarektE, 2020-09-29:
I translated I sent to Cedric PAILLE by e-mail

pakoto, 2020-09-29:
if you send here, i could create a full version install with it included to test.

BarektE, 2020-09-29:
[tr.zip](https://github.com/cedricp/ddt4all/files/5298504/tr.zip)

BarektE, 2020-09-29:
> Buraya gönderirseniz, test etmek için dahil edilmiş bir tam sürüm kurulumu oluşturabilirim.

gönderdim

BarektE, 2020-09-29:
gönderdim

BarektE, 2020-09-29:
> Buraya gönderirseniz, test etmek için dahil edilmiş bir tam sürüm kurulumu oluşturabilirim.

BarektE, 2020-09-29:
> / Locale içine açıp test edersiniz. Cedric'in tr dilini eklemek için ddt4all'ı değiştirmesi gerekip gerekmediğini bilmiyorum.
> 
> [tr.zip](https://github.com/cedricp/ddt4all/files/5293738/tr.zip)

pakoto, 2020-09-29:
i'm trying to convert. Some errors with " into messages, instead '

BarektE, 2020-09-29:
> dönüştürmeye çalışıyorum. Bunun yerine "mesajlara" ile ilgili bazı hatalar

gracias

pakoto, 2020-09-29:
https://mega.nz/file/XBsiiSoa#kKkbbF67ehvLwbuY46nHf278OHPb6SxW_O_xBK19iAI
Could you try to install?

BarektE, 2020-09-29:
> https://mega.nz/file/XBsiiSoa#kKkbbF67ehvLwbuY46nHf278OHPb6SxW_O_xBK19iAI
> Yüklemeyi deneyebilir misiniz?

estoy intentando

BarektE, 2020-09-29:
Turco al instalar pero después de instalar ddt4all no se abre

BarektE, 2020-09-29:
ddt4all no se abre después de la instalación

BarektE, 2020-09-29:
[image]

BarektE, 2020-09-29:
Probé en modo demo y déjame probar el auto

pakoto, 2020-09-29:
msgid "Diag version;Supplier;Soft;Version"
msgstr "Diag versiyonu; Tedarikçi; Yumuşak; Versiyon"

pakoto, 2020-09-29:
LETTER  Windows-1254    ISO-8859-9  latin5      UTF-8
ğ       b'\xf0'         b'\xf0'     b'\xf0'     b'\xc4\x9f'
ı       b'\xfd'         b'\xfd'     b'\xfd'     b'\xc4\xb1'
ş       b'\xfe'         b'\xfe'     b'\xfe'     b'\xc5\x9f'
ü       b'\xfc'         b'\xfc'     b'\xfc'     b'\xc3\xbc'
ö       b'\xf6'         b'\xf6'     b'\xf6'     b'\xc3\xb6'
ç       b'\xe7'         b'\xe7'     b'\xe7'     b'\xc3\xa7'
Ğ       b'\xd0'         b'\xd0'     b'\xd0'     b'\xc4\x9e'
Ü       b'\xdc'         b'\xdc'     b'\xdc'     b'\xc3\x9c'
Ş       b'\xde'         b'\xde'     b'\xde'     b'\xc5\x9e'
İ       b'\xdd'         b'\xdd'     b'\xdd'     b'\xc4\xb0'
Ö       b'\xd6'         b'\xd6'     b'\xd6'     b'\xc3\x96'
Ç       b'\xc7'         b'\xc7'     b'\xc7'     b'\xc3\x87'

BarektE, 2020-09-29:
> LETTER Windows-1254 ISO-8859-9 latin5 UTF-8
> ğ b'\xf0' b'\xf0' b'\xf0' b'\xc4\x9f'
> ı b'\xfd' b'\xfd' b'\xfd' b'\xc4\xb1'
> ş b'\xfe' b'\xfe' b'\xfe' b'\xc5\x9f'
> ü b'\xfc' b'\xfc' b'\xfc' b'\xc3\xbc'
> ö [...]

the car also works. Doesn't work in demo mode

BarektE, 2020-09-29:
I think the error of LETTER Windows-1254 ISO-8859-9 latin5 UTF-8 startup message is due to this.

pakoto, 2020-09-29:
only fails into dataeditor.py
I suppose that Cedric could fix it. 
i only can remove that line and use it in english

BarektE, 2020-09-29:
> yalnızca dataeditor.py'de başarısız olur,
> sanırım Cedric bunu düzeltebilir.
> sadece bu satırı kaldırıp ingilizce kullanabilirim

would be

BarektE, 2020-09-29:
Cedric PAILLE
Added to the GIT repository, Said she added

pakoto, 2020-09-29:
[tr.zip](https://github.com/cedricp/ddt4all/files/5298938/tr.zip)
modified only that line replacing Ç with C, could you test?

BarektE, 2020-09-29:
ok

BarektE, 2020-09-29:
Pakoto Thank you very much worked

pakoto, 2020-09-29:
now full works? i´ll packet again if you confirm it

BarektE, 2020-09-29:
> şimdi tam çalışıyor mu? Onaylarsan tekrar paketleyeceğim

şimdi tam olarak çalışıyor, olduğu gibi yeniden paketler misiniz?

pakoto, 2020-09-29:
https://mega.nz/file/LIVmzBJC#z6mK5POpATYOE6VX_q4DgQWS_14-osYn-_fBJ3mX9Ng
untouched DDT4all + 07.2020+tr language with line modified to bypass error.

BarektE, 2020-09-29:
pakoto thank you very much, works smoothly

BarektE, 2020-09-29:
[image]

BarektE, 2020-09-29:
there is problem. I haven't tried this place

pakoto, 2020-09-29:
older ddt4all version. try to open 4th tab
[image]

BarektE, 2020-09-29:
> eski ddt4all sürümü otomatik olarak 4. sekmeyi açar
> 
> [image]
> 
> [image]

it doesn't open up anywhere

pakoto, 2020-09-29:
i think that ecu don´t have "screen"
Renolink don´t show anything

[image]

BarektE, 2020-09-29:
[image]

[49 more replies]

closed 2020-09-30 by BarektE