cedricp/ddt4all #463: Turkish Translate

tanerdogan, 2020-09-30:
Hi,

As I said before, I am software developer and want to help about translate.

So I work on /locales/ddt4all.pot file and new file 
[dd4all_tr.po.zip](https://github.com/cedricp/ddt4all/files/5305668/dd4all_tr.po.zip)

I'be happy if someone merge...

Happy coding...

pakoto, 2020-09-30:
[tr.zip](https://github.com/cedricp/ddt4all/files/5305899/tr.zip)

pakoto, 2020-09-30:
works?

SedatAklar, 2023-01-05:
Taner bey şu ddt4all da mausun tekerleği yüzünden yanlışlıkla değişiklik sebebiyle dünyada bir çok kişinin cihazı bozuluyor. Sayfayı kaydırayım derken parametreler değişiyor ama hangisi değişti bunuda bilmek imkansız oluyor. Tekerlek çevirerek parametre değişimini iptal edebilirsen insanlar rahat edecek.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif