cedricp/ddt4all #465: The write button in ddt4all doesn't work.

jjaemny, 2020-10-02:
Hi, I have a some problem using DDT4all.

I know Using DDT4all is very simple.
1) turn on DDT4all
2) select my car
3) click "Einstein" and modification
4) click "write".

But.. I have a trouble.. 4) step.
After modificaion click "write", it is not save..

Those who succeed in coding say that it is possible to have an expensive product over $1,000, and they are charged 70 dollars each for coding. I feel this is unfair and try to spread the coding method to everyone.

i use ELM327, my car is XM3 UK named arkana.

below, error picture.
[image]
[image]

KarelSvo, 2020-10-04:
What's your VIN?

jjaemny, 2020-10-04:
My VIN is KNMK5B2RMLPxxxxxx

pilotkalem, 2020-11-04:
just connect can lines of the ELM directly to BCM can lines. OBD can lines are behind CGW - a secutiry gateway module so it will not allow you to WRITE ECU. BCM can lines are connected directly to INTERNAL NETWORK. 70$ is expencive for a basic cabling work :)

jjaemny, 2020-11-07:
> just connect can lines of the ELM directly to BCM can lines. OBD can lines are behind CGW - a security gateway module so it will not allow you to WRITE ECU. BCM can lines are connected directly to INTERNAL NETWORK. 70$ is expencive for a basic cabling work :)

Thank you for the good information. But is the bcm module compatible with elm327? From the naked eye, it is not possible to connect an elm327 module like an obd module.

jjaemny, 2021-05-19:
> What's your VIN?

Looking at the answer to the #404 issue and pilotkalem's answer, it seems that changes can be made by connecting ELM327 to BCM and running ddt4all.
But I don't know how to connect ELM327 to BCM.
Can you tell me how to connect ELM327 to BCM?
Thanks.

@pilotkalem @KarelSvo

bokkoman, 2021-06-10:
> > What's your VIN?
> 
> Looking at the answer to the #404 issue and pilotkalem's answer, it seems that changes can be made by connecting ELM327 to BCM and running ddt4all.
> But I don't know how to connect ELM327 to BCM.
> Can you tell me how to connect ELM327 to BCM?
> Thanks.
> 
> @pilotkalem @KarelSvo

I am wondering myself as well.
They tell us to just connect to BCM, but not how. 
If you succeeded in bypassing the security gateway, tell us exactly how.
Where do we connect? What cables? What connector? Where is the device located?

Please help fellow ddt4all users out 

@pilotkalem

AndyIreland, 2021-06-10:
> I am wondering myself as well.
> They tell us to just connect to BCM, but not how.
> If you succeeded in bypassing the security gateway, tell us exactly how.
> Where do we connect? What cables? What connector? Where is the device located?
> 
> Please help fellow ddt4all users out
> 
> @pilotkalem

would be good alright if we had some more info

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif