cedricp/ddt4all #468: renault r-link 2 downgrade ??

acgunner, 2020-10-05:
hello, ive installed software upgrade 9.9xxx into my 2016 renault kadjar, its not working, blew speakers, volume knob not working , phone connectivity not working, ect the list goes on !!
how do you downgrade back to the units original software 2.2.17.803 please ?? before i get a big bill off my local renault dealer??

AlMeu, 2021-08-01:
Big mistake.
Upgrade from 2.2 or 3.3 to 7.0 is making only at Renault dealers trough their servers.
It's a small chance to bring it back. If I have it on my desk I will give a try.

acgunner, 2023-03-08:
I still have my radio ,now covids over could i post it to you to see if you could save it for me ??

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif