cedricp/ddt4all #470: Renault Megane Mk3 Rlink - Vin write

Ghighi246, 2020-10-12:
Hello,

I have a renault megane mk3 with rlink retrofitted. When I check the settings in MFD ECU I saw that the VIN number is not inserted (there are a lot of 0).

Can I simply write the VIN number with ddt4all or EcuTweaker (Android)? Would it brick something?

Thanks in advance!

BarektE, 2020-10-12:
ELM327 CAN 2 portunu destekliyorsa VIN No: yazabilirsin.  Normalde ELM327 de Modifikasyon yapmanız gerekir.

Ghighi246, 2020-10-13:
I have a modified extender in order to access Multimedia CAN. My problem was that: can I brick the device if I write this VIN number?

fparmentier, 2020-10-31:
Yes you can write the VIN but there is not really a need to do so as far as I know.

Ghighi246, 2020-10-31:
Thank you all for your answers! Really helpful.

I want to write the VIN number because mai ecodrive function is not making any graphic calculations. For the moment the bord computer is shown correctly (distantce, consumption etc.) but the leaf is at 80/100 all the time, acceleration and anticipation graph is full and gear change is null. I don't understand what is set wrong on it.

On another forum I seen that someone mentioned that the crc number is used to make different ecu communicate accordingly. So I hope this crc number and vin write will solve my problema.

fparmentier, 2020-10-31:
Sounds interesting. I have also retrofitted RLink and don't have Eco drive working. Will change the VIN to see if I can make it work.

Ghighi246, 2020-10-31:
But the bord computer is transfering it's data from meter cluster to mfd?

fparmentier, 2020-10-31:
I have no idea

Ghighi246, 2020-10-31:
You can activate the ecodrive function from mfd settings with ddt4all. But if you have meter cluster before 2012 it will not work for sure

fparmentier, 2020-10-31:
Thanks, mine is from 2010. So I know what I have to do, buy a new cluster.

Ghighi246, 2020-11-04:
So, i managed to write cid but no update. Eco function still not calculate the acc gear and anticipation. Could be the bic?

fparmentier, 2020-11-05:
Maybe you have an old version of the BIC? I had do upgrade my BIC to make RS Monitor work, could be the same for eco functions?

Ghighi246, 2020-11-05:
Hy, when I replaced BIC I installed one with the following number 283467680R. When I search online I find this BIC as dismantled from Clio IV or Captur.

fparmentier, 2020-11-07:
I managed to find a new cluster (2012 analog). I installed it, changed the milage to match my car, did a rollback of the original configuratiton (PyRen) and changed the VIN in cluster and Mfd. Result: Eco functions not working.

Ghighi246, 2020-11-07:
Hy, what do you mean by analog? It was an RS version? Or a version from fluence (megane 3 from turkey)? 
Exactly what are you doing I want to do next week by borrowing a cluster from a friend. He has a higher mileage then me and I don't know how to proceed in order not to write his mileage into my abs ecu.

fparmentier, 2020-11-07:
As long as you don't drive with the car, the ABS and ECU won't get updated mileage (as far as I know!)
It was an GT version, I drive a Megane GT 180 2010 model and changed the cluster with a 2012 model. I updated the mileage by reading the eeprom from both clusters and copied the values from my old cluster onto my new cluster. After that wrote eeprom back. There are several ways to read eeprom, I did it with (genuine) Renolink.

Ghighi246, 2020-11-07:
Ok. I don't have renolink, just an elm327. I think I will try first with a different bic to see if there is anything changed... Then I will try with different cluster.
Maybe your cluster was on the limit with software version inside... Also, if your megane is with sport cluster, maybe you should set the bic to different script, for x95 RS

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836