cedricp/ddt4all #471: Restore or deactivate reverse buzzer.

JaimeEspinoz, 2020-10-13:
Hello everyone

I have a Megane III with parking sensors. The beep used to sound through the stereo, while showing in the R-Link screen.
I used _[Aide au parking] UPATransversal V2.0_valeo_ to change the level 'Buzzer Sound Index', only to find later that _buzzer_ isn't what my car has. Since then, I have no sound for parking sensors.

Is there a way to go back to the default values? Or can it be set in another place?

Thanks in advance.

Ghighi246, 2020-10-14:
Can you access multimedia CAN? There you can access the MFD and see what setting you have for UPA. Tomorrow or friday I will make a printscreen of the settings from my ParkingECU

JaimeEspinoz, 2020-10-15:
> Can you access multimedia CAN? There you can access the MFD and see what setting you have for UPA. Tomorrow or friday I will make a printscreen of the settings from my ParkingECU

Hi @Ghighi246 .
I don't think I can access multimedia CAN. I've just tried and I get "NO DATA" in every field.

[image]

But it's weird; I tried and I was able to deactivate and re-activate HFP.

Ghighi246, 2020-10-17:
Hello,

Of you have no data it means that you didn't make the connection properly in order to you a simple elm on multimedia can. Elm reads the can info from prins 6 and 14, but on renault the navi/radio/bic is on multimedia can which can be accessed on pins 12 and 13. You can search online for this modification of the elm device or use an obd2 extender where you replace the wires.

JaimeEspinoz, 2020-10-22:
Hi again.
I've made the modification. Now, when scanning for ECUs it only finds these two:
- [Audio] Radio 2012 v2.4
- [MIU] BIC/A2/A3 v24

If I try to _manually_ go to MFD I get:
Bad ELM response: WRONG RESPONSE : NR : Busy Repeat Request(7F3E21)

Any suggestion? 

Best

Ghighi246, 2020-10-23:
I can't think on something else, but you can try connect replace between them pins 12 and 13. Maybe there is a wrong connection in MFD.

gilbert250, 2023-02-12:
Hello i have same problem with No sound on pdc in rlink did you solve the problem?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif