cedricp/ddt4all #472: Ecu found but no relevant file

freeday01, 2020-10-15:
Hello

Can somebody help me with this Problem? The ecu is found but i can not operate on it.

What i am doing wrong? 

[image]

freeday01, 2020-10-16:
Nobody?

Marianpol, 2020-10-16:
NO DATA issuse is caused, because DDT4ALL is choosing wrong CAN protocol.
Laguna 2 PH2 has CAN 250kbps, but DDT4ALL selects CAN 500kbps.

[Here](https://github.com/cedricp/ddt4all/issues/415#issuecomment-658121266) I explained a bit about this error and how to resolve it.

freeday01, 2020-10-16:
I have changed this in the ecu.py file and now this happens [image]

Marianpol, 2020-10-16:
Now you have your connection with ECM is set up correctly and you are receiving a messages.
SubFunctionNotSupported for 3E (testerPresent service) is nothing that your really need to care about.

Can you receive any data successfully from any of your screens or you only got NO DATA values on any of them?

freeday01, 2020-10-17:
Thank you! Now it works! But It is not what i expected. 
Is there any chance to deactivate the second O2 Sensor in the ecu? Or did i have issuese to translate the french language

Marianpol, 2020-10-17:
Unfortunately, there is no way to disable the second O2 sensor using the DDT4ALL. It cannot read/write FLASH/EEPROM.

freeday01, 2020-10-17:
Thank you for your answer. Is there an Programm that can make ist happen? Do you know that?

cedricp, 2020-10-18:
You will save time by replacing the O2 sensor, the ECU program is designed to work with it. Or it may be more simple to send a valid (I guess 0.5V) voltage to the O2 sensor input of the ECU's so it doesn't detect of fault.

freeday01, 2020-10-18:
Okay thank you

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif