cedricp/ddt4all #474: Ddt4all dpf help

maxrom29, 2020-10-24:
Hi.  I want to thank the developers of this wonderful program (ddt4all).  I have a Renault laguna 2 ph2 1.9 dci 130 F9Q 758 and I am trying to do a forced regeneration with ddt4all but I don't know what values ​​I have to put in the 2 checkboxes.  I have tried pyren and I get that there is a scenario and I do not support it.  many thanks to those who can help me.
[image]

maxrom29, 2020-10-24:
[image]

maxrom29, 2020-10-24:
[image]

maxrom29, 2020-10-24:
Here

Marianpol, 2020-10-24:
PyRen don't support regeneration scenario but you can use its commands. Go to ECM tab (you can unlock this tab by sending a "cmd" command in the main module menu and refreshing a tab list by going into any tab and going back) and search for the VP040 PARTICLE FILTER REGENERATION command.

You want to do forced filter regeneration because you have encountered too many failed regenerations or the soot mass in the DPF has exceeded the spontaneous regeneration soot mass level?

I recommend to check the hoses that goes from the DPF to the differential pressure sensor. They like to crack or clog up which leads to problems with DPF regenerations (regeneration run to often and many of them are failed). Even if they look good at first glance, they can snap or fall apart when you try to bend them, so it's good to replace them anyway.

maxrom29, 2020-10-24:
Hi.  Thanks for your tips.  I did what you told me and managed to access the ECM menu and launch the parameter vp040 in demo mode.  Tomorrow I am going to test in the car and I will tell you.  The dpf I have it with 60G for a broken differential pressure cuff.  another thing.  in ddt4all that I have to put in the 2 boxes to do the regeneration?  Again thank you very much and sorry for the bad English since I use Google translator

Marianpol, 2020-10-24:
Yeah, in DDT4ALL you need to put two values into these boxes to start a regeneration process, but I don't know what kind of values should be inserted there. I've tried to investigate, what should go there, but with no luck. Unfortunately, I don't know what should go there, but command from DDT4ALL is a bit different, so I would stick to a verified command from a PyRen.

maxrom29, 2020-10-25:
thank you very much.  Anyway I will do it with the PyRen but I would like to know what values ​​are put in the boxes.  Let's see if one day it comes out.  thanks for everything

figs1, 2022-04-23:
> Hi. I want to thank the developers of this wonderful program (ddt4all). I have a Renault laguna 2 ph2 1.9 dci 130 F9Q 758 and I am trying to do a forced regeneration with ddt4all but I don't know [...]

Can anyone help me with ddt4all

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif