cedricp/ddt4all #475: Can I Set My Radio Clock with ddt4all ?

yavbuneaq, 2020-10-25:
Hi. I have a x38 renault fluence. I installed an afterstock radio (receiver or tape ? i don't know the exact word excuse me for my english)  to my car and can't change the radio clock because you can only set the clock via the original radio. The clock always flashes and it bugs me. I wan't to set it or close the display of clock. I'm curious if i can set it with ddt4all. I'm familiar with the program so if there is an option, if you can guide me through it i would appreciate it. Thank you and have a nice day

BarektE, 2020-10-25:
tatbiki ayarlarsınız

BarektE, 2020-10-25:
ELM 327 cihazınız CAN2 protokolünü destekliyorsa çok basit. 

[image]

yavbuneaq, 2020-10-26:
> ELM 327 cihazınız CAN2 protokolünü destekliyorsa çok basit.
> 
> [image]

Hocam bu menüye nasıl ulaşabilirim ? Girmediğim menü kalmadı yine de bu ayarı bulamadım. Teşekkür ederim

BarektE, 2020-10-26:
e- mail yazın oradan göndereyim. Ancak ELM 327 teybi desteklemesi şart

yavbuneaq, 2020-10-31:
> tatbiki ayarlarsınız

hocam bi süredir bakamıyorum buraya mail adresinizi öğrenebilir miyim menüye nasıl girebileceğim ile ilgili bilgi almak isterim. İyi günler

localhost61, 2020-11-08:
Did you look into "Meter Cluster" then "BICxxx" then "RDBI/WDBI" and finally "Configuration"

BarektE, 2020-11-09:
> > tatbiki ayarlarsınız
> 
> hocam bi ile bakamıyorum buraya mail adresinizi öğrenebilir miyim menüye nasıl girebileceğim ile ilgili bilgi almak isterim. İyi günler

E-postanızı yazın oradan nasıl yapacağınızla ilgili fotoğraf atarım

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif