cedricp/ddt4all #476: Macro commands

AlMeu, 2020-10-26:
Hy guys
I am trying build myself some macro for run and I don't understand something exactly how to calculate.
I found in some macro following sintax:

"
# check if it is rlink2
exit_if_not F18A 6 4 FFFFFF 434150

set_bits 2130 10 5 60 60
" 

First is how is identifying that the unit is really a R-Link2? what is interogating for get that answer **F18A 6 4 FFFFFF 434150** exactly? 
I want to interogate in same way to identify another unit type.

Second  is **set_bits** command. I understood the address **2130** where wants write but **10 5 60 60** I am not sure 100% how to calculate.  I know that **60** is the hex value for that area, but I don't get ideea of other numbers.

Thanks for help

shrlnm, 2020-10-27:
Hi
Look this with google translate
https://www.club-renault.ru/forum/topic/5528-diagnostika-svoimi-rukami/?do=findComment&comment=2234584

AlMeu, 2020-10-27:
Thanks a lot man.
It's exactly what I want.

shrlnm, 2020-11-01:
еще команды 
https://www.club-renault.ru/forum/topic/5528-diagnostika-svoimi-rukami/?do=findComment&comment=2271793

AlMeu, 2020-11-03:
Very usefull.

closed 2021-08-01 by AlMeu