cedricp/ddt4all #477: DDT4ll oroginal radio replacement

luissilva1971, 2020-10-29:
Hello.
I have a Megane 3 Grandtour 2010 with R01 radio (no bluetooth, ref. 281150030R) with carminat tomtom, and i would like to replace it by an original R02 radio (with bluetooth, ref. 281159389R) i bought from junkyard, that came from a car with no tomtom (standard disply instead), but keeping my tomtom operational.
Can it be done by ddt4all with ELM327 (bluetooth) with switch (change from H-CAN to L-CAN))? Anyone can give me a hand / tutorial how it can be done? I would be forever grateful.
Thanks in advance.

fparmentier, 2020-10-31:
Yes it will work. You will have to edit the radio and carminat settings aswell.
Both ecu's will appear if you do a scan. No need to know what xml is used. 
Don't forget the microphone is not fitted, you will have to retrofit that too. The wiring is already present (empty connector in right A pillar).

luissilva1971, 2022-03-17:
Hello, thank you for your reply. Search with ELM327 with CAN-2 (multimedia chosen with pinout switch), radio and navigation appear with "red" colour, with means that is not a exact macth between ECU xml and DDT4all database, right?
Manually i choose what? Radio i thinh is it the r208 v3.81, but for navigation? NFA v0.6? NFA2 V1.3? Could anybodu help? Thanks in advance. Kind regards.

MellotronScratch, 2022-03-17:
Take the one that is closest to the detected ecu.

luissilva1971, 2023-03-09:
Hello. I've achived to make the change with sucess. Now i have another thing i would like to do, is to update radio software version without using CLIP (but the radio dos not have a USB port...). :-(  Radio has 84.15 soft version, but the updated version is 88. Is there any way to do this without using CLIP software?
Thank you.

luissilva1971, 2023-03-14:
> As far as I know the only way to update is with Clip or with a USB stick. I don't know if it's worth the hassle to be honest.

it is because 8t vrrsion has e few bugs and random disconnects, version 88 corrects that. On Renault office shop they told me that because i ve changed radio and radio vin eith alternate software the clip software will not recognize the radio from renault database and blocks it. I kmow it can be done by "offline" versoon of clip but you need besides clip itselt (hardware and software) a software that returns a pin code in reply to a code request by clip. I do not have neither so....

MellotronScratch, 2023-03-14:
When connecting with clip you can fill in another VIN, so you can use the VIN of your new radio.
After that you can do reprogramming with no issues. It will only ask you what VIN it has to use, because it recognises two vins. I think you will need to find someone with clip or explain to the dealership what do to.

The pin codes can be generated with a custom tool.

luissilva1971, 2023-03-16:
Hello. So when connecting with clip i should introduce the vin from the original car? And then when doing the reprogramming it should be choseen my car vin?

MellotronScratch, 2023-03-16:
Almost. Only use the vin that's configured in the new radio. 
You can find the vin in clip, or with ddt4all.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif