cedricp/ddt4all #478: Can't use DDT4ALL on new Clio 5

lorenzomartini1, 2020-11-01:
I have a Clio 5 (2020), i'm trying to use the DDT4ALL for edit some parameters such as clock on display, ecc.
I have bought this ODB adapter which is a wifi one: https://www.amazon.it/gp/product/B01NAQ2878/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

I run DDT4ALL, i click the button next to car list to scan ECU (i think) and in bottom panel (where logs are) I see a lot of "Navi ECU found but no relevant file" or "Radio .........." red colored. At the end of scanning process I see only 4 green colored ECU and 1 red. Why so few?
If I change something on this ECU (founded), in log section I see: "_Send request to ELM: [hex code]_" and then "_ELM response_" always empty.

I surfed online for a solution and I found someone which says to click "ELM benchmark" to see if my ODB adapter is compatible. I did so and I see this in log window:

`Apertura porta COM192.168.0.10:35000 @ 38400
Traceback (most recent call last):
  File ".\ddt4all.py", line 1201, in check_elm
    res = elm.elm_checker(port, speed, self.logview, core.QCoreApplication)
  File "C:\Program Files (x86)\ddt4all\elm.py", line 1333, in elm_checker
    print cm[2] + " " + res.strip()
  File "C:\Program Files (x86)\ddt4all\Python27\lib\codecs.py", line 369, in write
    data, consumed = self.encode(object, self.errors)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xfc in position 47: ordinal not in range(128)
Inizializzazione ELM con velocit├á 38400...`

What I am doing wrong?

aldomic, 2020-11-03:
You are not doing wrong Doesn't connect to clio 5 with ddt4all.

lorenzomartini1, 2020-11-03:
> You are not doing wrong Doesn't connect to clio 5 with ddt4all.

so what should I do?

pilotkalem, 2020-11-04:
Clio V has security gateway module. You can not change parameters directly from OBD connector with DDT4ALL. Security module only allows to read parameters - thats all. All other requests blocks by security gateway.

Deswa16, 2021-02-08:
> Clio V, güvenlik ağ geçidi modülüne sahiptir. DDT4ALL ile doğrudan OBD konektöründen parametreleri değiştiremezsiniz. Güvenlik modülü sadece parametrelerin okunmasına izin verir - hepsi bu. Diğer tüm istekler güvenlik ağ geçidine göre engellenir.

How can he bypass the security module on the Clio 5

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif