cedricp/ddt4all #479: Changing Service interval with DDT4ALL

johnroberts2k, 2020-11-01:
I have a Megane MK3 and the factory programmed service interval is 18k miles / 24months. Does anyone know if there is a way to change the frequency using ddt4all so the service light comes on yearly or 10k miles ?

Thanks

tanerdogan, 2020-11-05:
Yes its possible but i dont remember location.

localhost61, 2020-11-08:
It may help you...
I have a Fluence MY2011 (L38) that shares lots of electronics with Megane 3.
I already did this with Renault CanClip in "Tableau de Bord" wich seems to be "Cluster" in English.
Looking in DDT4All Android port, If I select "Megane/Scenic III", then "Meter Cluster" if I choose "CLUSTER_L38_B32_X95RSGT" or "CLUSTER_L38_B32_X95RSGT_V6.0" I have access to the "MAINTENANCE CONFIGURATION" screen.

johnroberts2k, 2020-11-09:
> It may help you...
> I have a Fluence MY2011 (L38) that shares lots of electronics with Megane 3.
> I already did this with Renault CanClip in "Tableau de Bord" wich seems to be "Cluster" in English. [...]

Thanks, I will try that

localhost61, 2020-11-09:
... and remember to Close the Issue if you managed to find a solution ;-)

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif