cedricp/ddt4all #484: Error installing on windows 7 64bit

jrmarins, 2020-11-05:
When finishing the installation on windows using the installer (Release November 2020 Python 3 64bit) the following error appears:

[image]

And when trying to open the program, windows cannot find the application.

cedricp, 2020-11-05:
This is not an error, just a bug of the installer, you should read "don't forget to install database"
If it doesn't start, can you tell me the error, working for me.

jrmarins, 2020-11-05:
> This is not an error, just a bug of the installer, you should read "don't forget to install database"
> If it doesn't start, can you tell me the error, working for me.

Thank you reply.
When I try to open the application, this error appears:

[image]

Looking at the shortcut properties, under "destination" it looks like this:

"C:\Program Files (x86)\ddt4all\Python27\python.exe" "C:\Program Files (x86)\ddt4all\ddt4all.py"

cedricp, 2020-11-05:
shortcut is likely the old one, try to start from start menu or launch DDT4ALL.BAT in install dir

jrmarins, 2020-11-05:
using DDT4ALL.BAT opened, I can't test the connection now, but it seems to work normal.

closed 2020-11-06 by jrmarins