cedricp/ddt4all #485: Renault scienic 2 add card key

Forsmod, 2020-11-07:
Hello)
How can i add second card key to my Renault scienic 2 2007y? I found a way for scienic 2, <2006y. But this way didn't work on car 2007y.
I'll be very thankful for your reply))

abecikxp, 2020-11-07:
You need dump uch memory by mcu programmer for read pin to code new card

Forsmod, 2020-11-07:
I managed to read the code here. This is it?
[image]

Forsmod, 2020-11-07:
> You need dump uch memory by mcu programmer for read pin to code new card

I managed to read the code here. This is it?
<details>
  <summary>Image</summary>
[image]

</details>

cedricp, 2020-11-07:
ISK reading has been scrambled after 2006 (phase 2), you need to read it from EEPROM, but don't have more information.

Forsmod, 2020-11-07:
Ok, thank you :(

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif