cedricp/ddt4all #486: Remplacement radnav

sobaid13, 2020-11-08:
Bonjour j'aimerais savoir la procédure après un remplacement d autoradio sur Mégane 4 rad nav 3.3 en sachant que j en ai pris un d occasion merci

AlMeu, 2021-04-28:
It is quite complicate to remove on Megane 4. Better do on a workshop.
Changing the unit be carefull that is tricky. Unit must be without VIN and antitheft code, otherwise is useless.

sobaid13, 2021-04-28:
sàlut and for the survey the anti-theft code how I should do with which tools, and for the wine I put what in place thank you

Télécharger Outlook pour Android<https://aka.ms/AAb9ysg>

AlMeu, 2021-04-28:
Only VIN is not enough. When you will try install in car you will get blocked by antitheft code.
I can clean by VIN and antitheft code the unit, but I need it on my desk.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif