cedricp/ddt4all #488: Add new card on Laguna 2 ph 2 >2006

AssTuss, 2020-11-08:
I would like to have your opinion, I read that it is difficult to set up new cards for the laguna 2 ph2 (2007).

My cable after testing on ddt4all seems to be working fine (86 out of 95 positive responses) but I cannot learn a new badge. Can anyone help me? Do you have a detailed procedure (or a video) to help me?

Materials used:

- 4 in 1 FNR cable + "Chinese version truecode"
- ddt4all + ddt2000 (latest version I think ...)
- pine extractor 2
- Windows 7 computer

ISK of the car: 8F C5 75 E9 EC AE.
The pin that pin extractor gives me is AF 4F 65 74 7E 2B.
On ddt4all the pin does not seem to be accepted when I enter it the box turns red and notes me as invalid (I think it was this message)

Thanks for your help.

cedricp, 2020-11-08:
Retry with no spaces between hex values

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif